raspberrypi/piserver #4: Text highlighting on menus is unclear

jonwitts, 2018-01-05:
I have just installed the latest version from here and the highlighting of selected items is really unclear. See screen grab below. 
[image]

spl237, 2018-01-05:
That is due to whatever GTK theme you are using on your host machine, and specifically the GTK+3 theme which is associated with the root user account, as GTK+3 handles theming very badly (IMNSHO) when sudo is used.

Try installing the default PiX theme folder into /root/.themes/ on your machine.

jonwitts, 2018-01-05:
Thinking about this, it is probably linked to me having to downgrade two dependencies to get libgtkmm-3.0-dev to install.

I am going to revert the snapshot on my PiServer and re-run this so that I can get the exact details of what I had to downgrade for you...

spl237, 2018-01-05:
It's an issue I had in internal testing; I had to tweak the themes used by the root account to fix it. I don't think downgrading any packages would affect it, but various things about that screenshot (such as the font in use) suggest you do not have the correct GTK+3 PiX theme installed and functional.

spl237, 2018-01-05:
Ah - in fact, now I think about it, I patched the GTK+3 library to make it do sensible things with sudo applications. Check your version of libgtk-3-0; it needs to be 3.22.11-1+rpi3 to fix this.

jonwitts, 2018-01-05:
Before attempting to install the latest version of PiServer form GitHub all menus look OK.
[image]

Then on a fully updated RPi-desktop install, I get the following error when trying to install libgtkmm-3.0-dev:
`pi@PiServer:~ $ sudo apt-get install devscripts debhelper cmake libldap2-dev libgtkmm-3.0-dev libarchive-dev libcurl4-openssl-dev intltool
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtkmm-3.0-dev : Depends: libgtk-3-dev (>= 3.22.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`
So to try and fix that I install aptitude `sudo apt-get install aptitude` and run the following:
`pi@PiServer:~ $ sudo aptitude install libgtkmm-3.0-dev
The following NEW packages will be installed:
  autoconf{a} automake{a} autopoint{a} autotools-dev{a} debhelper{a} dh-autoreconf{a} dh-strip-nondeterminism{a} gettext{a} gir1.2-atspi-2.0{a} icu-devtools{a} intltool-debian{a} libarchive-zip-perl{a} libatk-bridge2.0-dev{a} 
  libatk1.0-dev{a} libatkmm-1.6-dev{a} libatspi2.0-dev{a} libcairo-script-interpreter2{a} libcairo2-dev{a} libcairomm-1.0-dev{a} libdbus-1-dev{a} libdrm-dev{a} libegl1-mesa-dev{a} libepoxy-dev{a} libfile-stripnondeterminism-perl{a} 
  libfontconfig1-dev{a} libgdk-pixbuf2.0-dev{a} libglib2.0-dev{a} libglibmm-2.4-dev{a} libgraphite2-dev{a} libgtk-3-dev{ab} libgtkmm-3.0-dev libharfbuzz-dev{a} libharfbuzz-gobject0{a} libice-dev{a} libicu-dev{a} libltdl-dev{a} 
  libmail-sendmail-perl{a} libpango1.0-dev{a} libpangomm-1.4-dev{a} libpcre3-dev{a} libpcre32-3{a} libpcrecpp0v5{a} libpixman-1-dev{a} libpthread-stubs0-dev{a} libsigc++-2.0-dev{a} libsigsegv2{a} libsm-dev{a} 
  libsys-hostname-long-perl{a} libtool{a} libwayland-bin{a} libwayland-dev{a} libx11-dev{a} libx11-doc{a} libx11-xcb-dev{a} libxau-dev{a} libxcb-dri2-0-dev{a} libxcb-dri3-dev{a} libxcb-glx0-dev{a} libxcb-present-dev{a} 
  libxcb-randr0-dev{a} libxcb-render0-dev{a} libxcb-shape0-dev{a} libxcb-shm0-dev{a} libxcb-sync-dev{a} libxcb-xfixes0-dev{a} libxcb1-dev{a} libxcomposite-dev{a} libxcursor-dev{a} libxdamage-dev{a} libxdmcp-dev{a} libxext-dev{a} 
  libxfixes-dev{a} libxft-dev{a} libxi-dev{a} libxinerama-dev{a} libxkbcommon-dev{a} libxrandr-dev{a} libxrender-dev{a} libxshmfence-dev{a} libxtst-dev{a} libxxf86vm-dev{a} m4{a} po-debconf{a} wayland-protocols{a} 
  x11proto-composite-dev{a} x11proto-core-dev{a} x11proto-damage-dev{a} x11proto-dri2-dev{a} x11proto-fixes-dev{a} x11proto-gl-dev{a} x11proto-input-dev{a} x11proto-kb-dev{a} x11proto-randr-dev{a} x11proto-record-dev{a} 
  x11proto-render-dev{a} x11proto-xext-dev{a} x11proto-xf86vidmode-dev{a} x11proto-xinerama-dev{a} xorg-sgml-doctools{a} xtrans-dev{a} 
0 packages upgraded, 100 newly installed, 0 to remove and 0 not upgraded.
Need to get 41.2 MB of archives. After unpacking 166 MB will be used.
The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libgtk-3-0 (= 3.22.11-1) but 3.22.11-1+rpi3~iso is installed
                Depends: gir1.2-gtk-3.0 (= 3.22.11-1) but 3.22.11-1+rpi3~iso is installed
The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     libgtk-3-dev [Not Installed]                       
2)     libgtkmm-3.0-dev [Not Installed]                   

Accept this solution? [Y/n/q/?] n
The following actions will resolve these dependencies:

     Downgrade the following packages:                                
1)     gir1.2-gtk-3.0 [3.22.11-1+rpi3~iso (now) -> 3.22.11-1 (stable)]
2)     libgtk-3-0 [3.22.11-1+rpi3~iso (now) -> 3.22.11-1 (stable)]    

Accept this solution? [Y/n/q/?] y
The following packages will be DOWNGRADED:
  gir1.2-gtk-3.0 libgtk-3-0 
The following NEW packages will be installed:
  autoconf{a} automake{a} autopoint{a} autotools-dev{a} debhelper{a} dh-autoreconf{a} dh-strip-nondeterminism{a} gettext{a} gir1.2-atspi-2.0{a} icu-devtools{a} intltool-debian{a} libarchive-zip-perl{a} libatk-bridge2.0-dev{a} 
  libatk1.0-dev{a} libatkmm-1.6-dev{a} libatspi2.0-dev{a} libcairo-script-interpreter2{a} libcairo2-dev{a} libcairomm-1.0-dev{a} libdbus-1-dev{a} libdrm-dev{a} libegl1-mesa-dev{a} libepoxy-dev{a} libfile-stripnondeterminism-perl{a} 
  libfontconfig1-dev{a} libgdk-pixbuf2.0-dev{a} libglib2.0-dev{a} libglibmm-2.4-dev{a} libgraphite2-dev{a} libgtk-3-dev{a} libgtkmm-3.0-dev libharfbuzz-dev{a} libharfbuzz-gobject0{a} libice-dev{a} libicu-dev{a} libltdl-dev{a} 
  libmail-sendmail-perl{a} libpango1.0-dev{a} libpangomm-1.4-dev{a} libpcre3-dev{a} libpcre32-3{a} libpcrecpp0v5{a} libpixman-1-dev{a} libpthread-stubs0-dev{a} libsigc++-2.0-dev{a} libsigsegv2{a} libsm-dev{a} 
  libsys-hostname-long-perl{a} libtool{a} libwayland-bin{a} libwayland-dev{a} libx11-dev{a} libx11-doc{a} libx11-xcb-dev{a} libxau-dev{a} libxcb-dri2-0-dev{a} libxcb-dri3-dev{a} libxcb-glx0-dev{a} libxcb-present-dev{a} 
  libxcb-randr0-dev{a} libxcb-render0-dev{a} libxcb-shape0-dev{a} libxcb-shm0-dev{a} libxcb-sync-dev{a} libxcb-xfixes0-dev{a} libxcb1-dev{a} libxcomposite-dev{a} libxcursor-dev{a} libxdamage-dev{a} libxdmcp-dev{a} libxext-dev{a} 
[... cut ...]
  x11proto-render-dev{a} x11proto-xext-dev{a} x11proto-xf86vidmode-dev{a} x11proto-xinerama-dev{a} xorg-sgml-doctools{a} xtrans-dev{a} 
The following packages are RECOMMENDED but will NOT be installed:
  libgtk-3-bin libgtk-3-bin:amd64 
0 packages upgraded, 100 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 44.2 MB of archives. After unpacking 166 MB will be used.
Do you want to continue? [Y/n/?] y
`
Accepting the downgrade allows the packages to install and then allows me to install the rest of the dependencies to compile PiServer. However menu highlighting is now as shown below:
[image]

jonwitts, 2018-01-05:
Sorry, markdown managed by post there!

So the issue is certainly with libgtk-3-0 but I had to downgrade it to version 3.22.11-1 (stable) to get libgtkmm-3.0-dev to install. 

Can I force it back to the previous version now that libgtkmm-3.0-dev is installed?

spl237, 2018-01-05:
I'd have thought so - I'm not even sure why it had a problem with that version of libgtk-3-0 anyway, as the dependencies suggest it should be fine. Give it a try and see what happens.

jonwitts, 2018-01-05:
Umm... so having difficulty finding the rpi3 version of the package now with aptitude...

pi@PiServer:~ $ sudo aptitude install libgtk-3-0=3.22.11-1+rpi3
Unable to find a version "3.22.11-1+rpi3" for the package "libgtk-3-0"
Unable to apply some actions, aborting

spl237, 2018-01-05:
Try with apt rather than aptitude, and make sure you have an /etc/apt/sources.list.d/raspi.list file.

jonwitts, 2018-01-05:
No joy there either...

pi@PiServer:~ $ sudo apt install libgtk-3-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgtk-3-0 is already the newest version (3.22.11-1).
libgtk-3-0 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@PiServer:~ $ sudo apt install libgtk-3-0=3.22.11-1+rpi3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '3.22.11-1+rpi3' for 'libgtk-3-0' was not found

spl237, 2018-01-05:
What does "apt-cache policy libgtk-3-0" report?

jonwitts, 2018-01-05:
pi@PiServer:~ $ apt-cache policy libgtk-3-0
libgtk-3-0:
  Installed: 3.22.11-1
  Candidate: 3.22.11-1
  Version table:
 *** 3.22.11-1 500
        500 http://10.0.96.92:3142/ftp.debian.org/debian stretch/main i386 Packages
        100 /var/lib/dpkg/status

spl237, 2018-01-05:
OK, you aren't seeing our apt repo, so you can't see our version of the file. You should have a raspi.list file in /etc/apt/sources.list.d/ - I suspect you haven't. Go back to the clean image, copy that file somewhere safe, and put it back after installing libgtkmm; then do an apt-get update, and apt should then see our version of the file.

jonwitts, 2018-01-05:
OK, strange... Deffiently have the raspi.list in my sources.

pi@PiServer:~ $ cat /etc/apt/sources.list.d/raspi.list 
deb http://10.0.96.92:3142/archive.raspberrypi.org/debian/ stretch main ui
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ stretch main ui

The change to the url is because we are pulling all of our apt requests through an apt-cacher-ng server in school.

Do you mean to back up the libgtk file or the source list?

spl237, 2018-01-05:
I meant to back up the raspi.list file. Something odd is happening in your setup - when you do apt-cache policy libgtk-3-0, you should see a line referencing our server (archive.raspberrypi.org), but you aren't. This may be some side effect of running apt-cacher-ng, but as I can't repeat the problem here, I can't really help fix it - I can only suggest connecting your machine somewhere it can see our server directly without having to go through your local cache.

Out of interest, what is the result of apt-cache policy libgtk-3-0 on a clean install in your network? Can you see our server then?

jonwitts, 2018-01-05:
OK, have just reverted the snapshot. Here is the output of apt-cache:

pi@PiServer:~ $ apt-cache policy libgtk-3-0
libgtk-3-0:
  Installed: 3.22.11-1+rpi3~iso
  Candidate: 3.22.11-1+rpi3~iso
  Version table:
 *** 3.22.11-1+rpi3~iso 100
        100 /var/lib/dpkg/status
     3.22.11-1 500
        500 http://10.0.96.92:3142/ftp.debian.org/debian stretch/main i386 Packages

jonwitts, 2018-01-05:
I can open the firewall up so that the server can connect directly and take apt-cacher out of the equation...

spl237, 2018-01-05:
That might be the right thing to do - if you have a valid raspi.list (which it looks as if you do), and our server appears to be up and has the right file on it (I've just checked), it looks as if it is a network issue of some sort.

maxnet, 2018-01-05:
>Ah - in fact, now I think about it, I patched the GTK+3 library to make it do sensible things with sudo applications. Check your version of libgtk-3-0; it needs to be 3.22.11-1+rpi3 to fix this.

Please also make a +rpi3 version of libgtk-3-0-dev available on the repository.
Dev packages must be exact same version of normal version.

spl237, 2018-01-05:
There is already a +rpi3 version of the dev package on the repo - http://archive.raspberrypi.org/debian/pool/ui/g/gtk+3.0/

jonwitts, 2018-01-05:
So just the same issues with a direct apt connection.

Once I downgrade the two files to the standard Debian version I can no longer reinstall the +rpi3 versions...

jonwitts, 2018-01-05:
Is it a +rpi3 version of libgtkmm-3.0-dev which is needed?

maxnet, 2018-01-05:
>There is already a +rpi3 version of the dev package on the repo

Well, my apt has problems finding it.

==
pi@raspberrypi:~ $ sudo apt-get install devscripts debhelper cmake libldap2-dev libgtkmm-3.0-dev libarchive-dev libcurl4-openssl-dev intltool
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtkmm-3.0-dev : Depends: libgtk-3-dev (>= 3.22.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

==

pi@raspberrypi:~ $ sudo apt-get upgrade libgtk-3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libgtk-3-0 (= 3.22.11-1) but 3.22.11-1+rpi3~iso is to be installed
                Depends: gir1.2-gtk-3.0 (= 3.22.11-1) but 3.22.11-1+rpi3~iso is to be installed
                Depends: libglib2.0-dev (>= 2.49.4) but it is not going to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.37.3) but it is not going to be installed
                Depends: libatk1.0-dev (>= 2.15.1) but it is not going to be installed
                Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
                Depends: libepoxy-dev (>= 1.0) but it is not going to be installed
                Depends: libfontconfig1-dev but it is not going to be installed
                Depends: libx11-dev but it is not going to be installed
                Depends: libxext-dev but it is not going to be installed
                Depends: libxinerama-dev but it is not going to be installed
                Depends: libxi-dev but it is not going to be installed
                Depends: libxrandr-dev but it is not going to be installed
                Depends: libxcursor-dev but it is not going to be installed
                Depends: libxfixes-dev but it is not going to be installed
                Depends: libxcomposite-dev but it is not going to be installed
                Depends: libxdamage-dev but it is not going to be installed
                Depends: libegl1-mesa-dev but it is not going to be installed
                Depends: libwayland-dev (>= 1.9.91) but it is not going to be installed
                Depends: wayland-protocols (>= 1.7) but it is not going to be installed
                Depends: libxkbcommon-dev but it is not going to be installed
E: Broken packages
==

Had this problem earlier as well, when I was developing the csv import code.
(Worked around it by building under Ubuntu)

spl237, 2018-01-05:
OK, looks as if we may have some missing entries in the indexes on the server, so although the files are there, they aren't being found by apt. I'll get our server guys to look at that.

jonwitts, 2018-01-05:
Awesome! Packages all installing without issue now :-)

closed 2018-01-05 by jonwitts