raspberrypi/piserver #5: Option to add users to sudoers file.

mverleun, 2018-01-05:
When a user logs in he/she is not listed in the sudoers file.
Please provide an option to add them, or make them member of a group that is allowed to use sudo

ghollingworth, 2018-01-05:
Click on the 'software' then 'shell'

In the shell do

vi /etc/sudoers

Add a line for the user you want to have sudo access (i.e. copy and paste the root= and change)
Save it

mkdir /var/lib/sudo/ts

Add a tmpfs for /var/lib/sudo/ts to /etc/fstab

That will work for you (worked for me...)

maxnet, 2018-01-05:
What is the exact problem you are trying to solve by making users part of the sudo group?

Note that even if users have sudo permissions, they will not be able to install extra software and such that way, as the root file system is read-only.
So its usefulness is pretty limited.

If you need to install extra software, use the 'software' -> 'shell' method.

mverleun, 2018-01-05:
It’s not really a problem I’m trying to solve, but the question is from a different perspective.

When using the Raspberry Pi to teach Linux Essentials students sometimes have to use the sudo command.
I realize that there are no permanent settings that they can change, it’s just for the exercise of being able to do it.

E.g. it would allow them to create a file owned by root in their homefolder and then be surprised to see that they can remove it as a regular user. I use this to explain how the permissions work.

maxnet, 2018-01-05:
>E.g. it would allow them to create a file owned by root in their homefolder and then be surprised to see 
>that they can remove it as a regular user. I use this to explain how the permissions work.

Would not work either.
Home folder is mounted through sshfs, that does not support changing ownership of files.

mverleun, 2018-01-05:
Didn’t realise that sshfs doesn’t support changing owners. 
I guess I’ll continue to use sd cards for the class.

Thanks for your replies.

closed 2018-01-11 by ghollingworth