raspberrypi/piserver #6: Installing on Ubuntu

chrisdecker1201, 2018-01-06:
Hallo,

I've tried to build & install it on Ubuntu 16.04 LTS because I thought it'll work.
During the building and installation, I had several issues.

1. missing dependencies: qemu-user-static gksu. I had to install them, fix it with _sudo apt -f install_ and rebuild
2. The last outputs after the building process was
`Finished running lintian.
Now signing changes and any dsc files...
 signfile piserver_0.1.dsc Floris Bos <bos@je-eigen-domein.nl>
gpg: »Floris Bos <bos@je-eigen-domein.nl>« übersprungen: Geheimer Schlüssel ist nicht vorhanden
gpg: /tmp/debsign.NXiSjT3j/piserver_0.1.dsc: clearsign failed: Geheimer Schlüssel ist nicht vorhanden
debsign: gpg error occurred!  Aborting....
debuild: fatal error at line 1295:
running debsign failed
`
3. after starting with _sudo piserver_ a get the following but every seemed to work
`
(piserver:29770): GLib-GObject-CRITICAL **: g_value_type_transformable: assertion 'G_TYPE_IS_VALUE (src_type)' failed
(piserver:29770): Gtk-WARNING **: /build/gtk+3.0-2Ut_nl/gtk+3.0-3.18.9/./gtk/gtkliststore.c:836: Unable to convert from (null) to gchararray
(piserver:29770): GLib-GObject-CRITICAL **: g_value_type_transformable: assertion 'G_TYPE_IS_VALUE (src_type)' failed
(piserver:29770): Gtk-WARNING **: /build/gtk+3.0-2Ut_nl/gtk+3.0-3.18.9/./gtk/gtkliststore.c:836: Unable to convert from (null) to gchararray
(piserver:29770): GLib-GObject-CRITICAL **: g_value_type_transformable: assertion 'G_TYPE_IS_VALUE (src_type)' failed
(piserver:29770): Gtk-WARNING **: /build/gtk+3.0-2Ut_nl/gtk+3.0-3.18.9/./gtk/gtkliststore.c:836: Unable to convert from (null) to gchararray
(piserver:29770): GLib-GObject-CRITICAL **: g_value_type_transformable: assertion 'G_TYPE_IS_VALUE (src_type)' failed
(piserver:29770): Gtk-WARNING **: /build/gtk+3.0-2Ut_nl/gtk+3.0-3.18.9/./gtk/gtkliststore.c:836: Unable to convert from (null) to gchararray
`

4. after I select a raspberry pi, user and image I clicked on Install and get the following and PiServer will be aborted:
`
Signing certificate...
SASL/EXTERNAL authentication started
SASL username: gidNumber=0+uidNumber=0,cn=peercred,cn=external,cn=auth
SASL SSF: 0
modifying entry "cn=config"
Failed to start piserver_ssh.service: Unknown unit: piserver_ssh.service
See system logs and 'systemctl status piserver_ssh.service' for details.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
`

Thank you very much for your support.

maxnet, 2018-01-06:
At the risk of offending GTK+ fans: GTK-warnings can be safely ignored.
They are more often bugs in GTK itself rather than in the application using it.

In this instance they occur when loading our user interface files (data/*.glade)
Well, I did not hand draft those in a text editor, but created them with Glade, a tool created by the GTK guys...

==

The error that you do not have my code sigining key when running debuild is normal as well.
Can install the .deb file produced regardlessly.

==

Can you try if executing "systemctl daemon-reload" solves the problem of systemd not finding the piserver_ssh unit?

==

You will also need to compile dnsmasq from source, or borrow packages from a newer Ubuntu server.
The version shipped with Ubuntu 16.04 is from 2015 and doesn't have our patches.

jmccoy555, 2018-03-31:
It works on the latest 18.04 build. I did a minimal install (which is very nice an light, just what you want for a server) and then dpkg -i the .deb package. It will want a few dependencies but they were all available. 

Only observation that didn't work is the 'Shell' button in the GUI, but you can just chroot in manually.

maxnet, 2018-03-31:
Ah, shell button is currently hard coded to start `lxterminal`
So indeed won't work if that is not installed.
May be better to let it fallback to something like `xterm`

maxnet, 2018-03-31:
Fixed in 6a95e708cb2f1eb74ae235cb661551bd546cfa46

closed 2018-03-31 by maxnet