raspberrypi/piserver #7: Problem with dnsmasq

ADTL, 2018-01-09:
Normally build on Debian 9 and Ubuntu 16.04, but after find and add some RaspberriPi
in console was written:
```
Job for dnsmasq.service failed because the control process exited with error code. See "systemctl status dnsmasq.service" and "journalctl -xe" for details.
```
Try 
```bash
>PiServ$ systemctl status dnsmasq.service

● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: failed (Result: exit-code) since Вт 2018-01-09 22:54:48 CET; 10min ago
  Process: 86916 ExecStop=/etc/init.d/dnsmasq systemd-stop-resolvconf (code=exited, status=0/SUCCESS)
  Process: 87018 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=1/FAILURE)
  Process: 87014 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 85948 (code=exited, status=0/SUCCESS)

янв 09 22:54:48 UbntRaspi systemd[1]: Stopped dnsmasq - A lightweight DHCP and caching DNS server.
янв 09 22:54:48 UbntRaspi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
янв 09 22:54:48 UbntRaspi dnsmasq[87014]: dnsmasq: syntax check OK.
янв 09 22:54:48 UbntRaspi dnsmasq[87018]: dnsmasq: extraneous parameter at line 7 of /etc/dnsmasq.d/piserver
янв 09 22:54:48 UbntRaspi systemd[1]: dnsmasq.service: Control process exited, code=exited status=1
янв 09 22:54:48 UbntRaspi systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
янв 09 22:54:48 UbntRaspi systemd[1]: dnsmasq.service: Unit entered failed state.
янв 09 22:54:48 UbntRaspi systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
```
And try 
```bash
>PiServ$ /etc/init.d/dnsmasq systemd-exec

dnsmasq: extraneous parameter at line 7 of /etc/dnsmasq.d/piserver
```
line 7 is "tftp-unique-root=mac"
seems that parameter "mac" not supported

Version dnsmasq in Ubuntu 16.04: 
```bash
>PiServ$ dnsmasq -v
Версия dnsmasq 2.75  Copyright (c) 2000-2015 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
```
Version dnsmasq in Debian 9:
```bash
>PiServ$ dnsmasq -v
Dnsmasq version 2.76  Copyright (c) 2000-2016 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
```

maxnet, 2018-01-09:
We had to add some extra functionality (tftproot per MAC) and a workaround for a bug in the bootrom (dhcp-reply-delay option) to dnsmasq to make things work the way we wanted them to.
Although we contributed the changes back to the upstream dnsmasq project, the Linux distributions you quote are too old to have them.

Simply put: once a new Ubuntu release gets out, they no longer update software to newer versions.
The updates you get when you do "apt-get upgrade" are security and stability fixes backported to the software version they set in stone.
They do not allow updates to add new functionality.

So while using Ubuntu's "long term support" 16.04 release gives you the benefit that you have stable software that is supported for 5 years, it also means you are stuck with software functionality from 2016 (or even earlier) during all that time...

Your options are:

- compiling dnsmasq from source.
- borrow the dnsmasq packages from a newer non-LTS Ubuntu release such as 17.10
- borrow the dnsmasq packages from the Raspbian x86 repository. Although that runs an older dnsmasq version as well, it has our patches backported to it.
- running Raspbian x86 in a virtual machine under Ubuntu

ADTL, 2018-01-10:
Confirming, under Raspbian x86 all is working.

closed 2018-01-10 by ADTL