raspberrypi/piserver #8: Add Epoptes to the PiServer

AsmoKoskinen, 2018-01-11:
Add Epoptes to the PiServer.
http://www.epoptes.org/

It works, but it might needs some tweaks. I did change static ip for the PiServer and added a line to the etc/hosts file.

`root@raspberry:/# cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	raspberrypi
192.168.100.100 server`

`root@raspberry:/# epoptes-client -c
depth=0 C = AU, ST = Some-State, O = Internet Widgits Pty Ltd
verify error:num=18:self signed certificate
verify return:1
depth=0 C = AU, ST = Some-State, O = Internet Widgits Pty Ltd
verify return:1
DONE
Successfully fetched certificate from localhost:789`

https://www.raspberrypi.org/forums/viewtopic.php?f=116&t=202184

Best Regards Asmo Koskinen.

maxnet, 2018-01-11:
I actually did play with Epoptes during testing, but it had some issues, which would need to be resolved first.
In particular:

* Did not work properly when no user is signed-in to the Pi.
* Epoptes server process crashed when logged-in user logs out.
* Some options like full screen broadcast are broken (windowed did work)

As far as the IP goes: dnsmasq acts as DNS server for all clients, and we make the piserver IP-address available under hostname "piserver"
Can tell Epoptes to use that by adding an option to /etc/default/epoptes-client

```
SERVER=piserver
```

More reliable than patching /etc/hosts as piserver IP can change as well (we do not require users to set static IP).

AsmoKoskinen, 2018-01-11:
I can live in not so perfect world ;-) But fair enough, if somebody needs Epoptes, it is quite easy to install after following installation from Epoptes site ("Client package installation for standalone clients").

Thank you for answering. Best Regards Asmo Koskinen.

ghollingworth, 2018-01-11:
Think it would be worth trying to get these problems fixed / worked around, it's a very useful GUI for controlling a bunch of clients.  We could then add the epoptes-client into the default PiServer client and add the epoptes server to the standard x86 and Raspbian images

gbaman, 2018-01-17:
Having had Epoptes shipped with PiNet for number of years now, I can confirm the software is just simply buggy unfortunately. It is a great package and a lot of the time works flawlessly, frequently for example though a single client will be missing, or as mentioned full screen broadcast doesn't work.
As well, watch out for one I could never trace back, which involves when you lock a machine, in some rare cases you can then never actually unlock it after...

So be warned, an excellent piece of software, but the versions shipped with Raspbian/Ubuntu both are very much full of bugs that always seem rather random... 
I have plenty of educators that have mentioned its unreliability.

maxnet, 2018-01-17:
The full screen broadcast issue was relatively easy to trace. Fix is in the "Do not pass -UseLocalCursor=0 to xtigervncviewer" PR referenced above.
Broadcasts work by starting a VNC viewer on the client computer.
Since the VNC viewer installed can differ depending on Linux distribution they support 3 different ones.
xtigervncviewer -which we have- supports command line arguments that are very similar to the others, but not the "-UseLocalCursor=0" argument, which caused it to error out.
Problem with bugs like this is that they may not show for other developers who use another Linux distribution that has a different vnc client, and therefore may go unnoticed for a long time.

Anyway, I believe those small issues are fixable, however I do have my concerns about the long term sustainability of supporting the software.
Note that upstream's last formal release was in 2016, development doesn't seems to be so active, and it has dependencies that are aging.
E.g.

* uses pygtk, which is unmaintained and [scheduled to be removed in the next Debian release](https://lists.launchpad.net/epoptes/msg01323.html).
* is written in Python 2, scheduled to go the same way in 2020

That all may seem to be far away, but I do believe it needs some thought, if we want to add Epoptes to piserver by default.

cbenhagen, 2018-06-18:
@maxnet You might be interested in the GSoC2018 project with will address most if not all of your concerns:
https://github.com/eellak/gsoc2018-epoptes

maxnet, 2018-06-30:
Other potential alternative -if there is any demand- could be Veyon: https://veyon.io/
Wouldn't have recommended it before, as it pulls in heavy weight dependencies (Qt 5).
But we now already have those for qpdfview.