raspberrypi/piserver #9: Strict sequence on initial use

m214089, 2018-01-15:
Hi,
I really like piserver and think that the choice of tools is very good. The only issue I have is that I want to start setting up the user base and software first and than move the server to a place where all the Pis are available. But that doesn't work. For schools the possibility to have different user groups in LDAP would be as well a quite nice thing. 
Anyhow, excellent decision to provide piserver to community!
Thanks a lot,
Luis

maxnet, 2018-01-16:
* Can start with a single Pi at home first, and add more Pi later (in main interface "clients" -> "add")

* What purpose do you suggest groups have? Just organization to make it easier to manage users in the user-interface? Or to have shared folders and similar functionality?

m214089, 2018-01-17:
Hi Maxnet,

well, the first might be my own laziness. I'm running a system which looks similar to one with piserver, but managed by 'hand' since four years. And all the PIs which would be involved are incorporated in a pupil build custom case and hard to get out and all information for dhcp and tftp are already available. So this would  be a convenience feature only.

The groups are more important. We have created groups for teachers and a kind of school year based system, so that we easily can remove accounts when kids leave school. A second level of the groups is related to create Mailing lists as we have projected oriented groups, which are living for some time only. We use  LDAP as well to feed a blog server, and a git repository server. Beside the missing group I miss an entry column for the email too. 
Cheerio,
Luis

maxnet, 2018-01-17:
>We use LDAP as well to feed a blog server, and a git repository server.

So you already have a server in your network that has all the pupil information.
Wouldn't it then not be easier if we added an option to let Piserver authenticate users against your existing Active Directory or other LDAP server?

m214089, 2018-01-17:
Hi Maxnet,
yes, we have one and, of course, and having the proposed added option would be a solution. I was wondering in how far this additional possibilities this could be of good use for other groups in the mid-term .  I would be willing to add some work to this, but I'm very busy at the moment. In conclusion: for us it would be fine to have the possibility to query another LDAP server, but in the 
long term it would be good to get the projects and email added. I know a couple of other teachers which just didn't start a project, because they do not have an idea on how to and would need a piserver kinf of thing.
I really appreciate your interest!
Luis

maxnet, 2018-01-17:
>I know a couple of other teachers which just didn't start a project, because they do not have an idea on how to and would need a piserver kinf of thing.

What setup do those currently use for normal computers?
Do they just have a bunch of standalone computers that do not require any password to use?

Or do students login with an individual username or password after all?
While I understand that not every teacher may start a project such as yours, I am just wondering if most schools aren't running LDAP servers already, without them realizing it.
Note that Active Directory which one gets if you install Windows Server and create users there, speaks LDAP as well.

m214089, 2018-01-17:
Most use a bunch of computers (mainly laptops) with one anonymous pupil account. Every once a while the laptops get newly installed to remove the left over chaos (once or twice a year).

There is a school district wide software solution available, but that's a closed solution without interfaces. All servers are run by a contractor of the district. This system has some educational components like personalized wifi access and virtual classrooms. In essence it only moves the paper sheets from the table into a cloud type of thing. And there is no chance to use that system.

maxnet, 2018-01-17:
Is that closed solution really cloud based to the point that if your Internet connection is down, you can no longer login on the computers at all?

Some of the school management solutions out there are essentially active directory front-ends, and are more open than they may appear on first sight.
The fancy management webinterface they provide do are hosted by the vendor. But if you tell it to create a user in the webinterface, it actually nudges a daemon the contractor installed on the Windows Server within the school premises to create a user locally.
If that's the case, we could detect and use that local server as well.

That said.
I do understand the desire for more open solutions.

m214089, 2018-01-17:
Network down no login anymore. However, that hardly happens as the city sponsors/maintains its own network (fiber to and in the building) and service is on industrial level. We can only access and do changes in the router and the associated app.

m214089, 2018-01-17:
Well, we can still enter all machines we maintain ourself. Which includes the before mentioned set of laptops.

maxnet, 2018-01-22:
>yes, we have one and, of course, and having the proposed added option would be a solution.

[image]

As an experiment, added some very minimal support for external LDAP/AD servers.
Adds a page to the installation wizard.
If user selects he wants to use external server, it will attempt to auto-detect domain (by asking DHCP server), and AD/LDAP server address (by looking up _ldap_tcp.domain SRV DNS record).
User then just needs to enter the login details (if required by the LDAP server), and press "next" to establish the link.
Most environments allow logging into LDAP with any normal user account (administrative access not necessary)
So may even work in environments in which the infrastructure is maintained by external parties.
Although I have not tested that.

m214089, 2018-01-23:
Hi maxnet,

cool, I could access my own LDAP server without problems. The official LDAP one is behind a Shibboleth 2.0 based SSO federation. So I could not use that one. Anyway, this is already 
a good step forward.
Thanks,
Luis

closed 2018-06-01 by m214089