raspberrypi/piserver #11: How to enable autologin in the client piserver

userraspberry, 2018-01-23:
How to enable automatic logon in the system on the raspberry pi.

maxnet, 2018-01-23:
What is the exact use-case?

1) Do you have a need to be able to save files to your home directory and have them persist?
And if so, what applications do you plan to use exactly?

Not all applications can deal with the situation that multiple Pi are logged into the same user account and write temporary files to the same home directory simultaneously.

2) Or do you intend to use Pi for Internet kiosk style purposes, in which there is no need to store anything?

That is easier to achieve.

userraspberry, 2018-01-23:
2. There is no need to save the files.

maxnet, 2018-01-23:
The following should work in theory.
You will have to test how well it works in practice for your exact purposes though.

In the Piserver GUI, go to "software" tab, select "Raspbian" and press "shell" button to get a chroot shell.

Disable mounting of the home folder on the central server, by renaming the pam_mount configuration file to something else:

```
mv /etc/security/pam_mount.conf.xml /etc/security/pam_mount.disabled
```

Add a local pi user:

```
adduser pi
```

Edit /etc/fstab

```
leafpad /etc/fstab
```

Modify:

```
tmpfs           /home           tmpfs   defaults,mode=755            0       0
```

To:

```
tmpfs           /home/pi           tmpfs   defaults            0       0
```

Save the file, and close leafpad
Start raspi-config

```
raspi-config
```

Enable automatic login there. ("boot options" -> "desktop/cli" -> "desktop autologin")

Be aware that in this setup `/home/pi` will live in tmpfs (ramdisk), and is entirely empty on each boot.
If you need to autostart any applications do so using global configuration directories, e.g. by creating a .desktop file in `/etc/xdg/autostart`
and not by putting files in home directory as they will not be seen.

jmccoy555, 2018-03-26:
Hi, this works great, however I need to persist data between boots. Particularly a selection made in Chromium running in kiosk mode. Is it possible to map the home folder to somewhere? 

Maybe making use of #15 although I note that this is also read only, which would be fine if there was a way for an 'admin user' with write access to initially set this?

maxnet, 2018-03-26:
>Is it possible to map the home folder to somewhere?

Yes, but:

- Sharing a single home directory with multiple computers read-write is problematic.
- Having a read-only home folder is problematic

>Maybe making use of #15 although I note that this is also read only, which would be fine if there was a 
>way for an 'admin user' with write access to initially set this?

Put your files in the shared folder prepared on the piserver computer, and create a one-line `/etc/rc.local` script that populates /home/pi on boot?

```
#!/bin/sh

cp -a /mnt/shared/preparedfiles/* /home/pi
```

closed 2018-04-13 by maxnet

map7, 2019-03-29:
I've found that the autologin might not be working due to a timing issue. When the login screen appears if I type my username and password in really quickly (5seconds) then it fails everytime. If I wait 10seconds then hit enter, it works. 

Could this be why autologin isn't working?

maxnet, 2019-04-02:
>I've found that the autologin might not be working due to a timing issue. 

That is unrelated.

As stated before, you do can enable autologin, but only if you do NOT need writable storage.
If you need writable storage, the piserver server expects a password from the client. And autologin does not provide any.

Buzz2912, 2019-12-05:
Just installed buster server und tried to enable autolgin the way steted above.
Using raspi-config to enable desktop login fails /bin/cat: /proc/device-tree/model: no such file or directory

maxnet, 2019-12-05:
>Using raspi-config to enable desktop login fails /bin/cat: /proc/device-tree/model: no such file or directory

That message should not affect your changes.
/proc/device-tree/model is only consulted to mention the Raspberry Pi model running in the title bar.