raspberrypi/piserver #12: Error Installing Raspbian

rahul-thakoor, 2018-01-25:
Hello, I am getting this error when trying to install Raspbian from the repository:

`Error executing 'certtool --generate-self-signed --load-privkey /etc/ldap/piserver.key --template=/etc/ldap/piserver.tpl --outfile=/etc/ssl/certs/piserver.pem'`

Any fix?
Thanks

maxnet, 2018-01-25:
What is the exact error message you get when you execute that command manually from a shell as root (prepend sudo)?

rahul-thakoor, 2018-01-25:
Thanks for the prompt reply.

```
Generating a self signed certificate...
Overflow while parsing days
```
Should I try modifying the `/etc/ldap/piserver.tpl` file?

maxnet, 2018-01-25:
Ok, will get that fixed.
We secure communication between Piserver and clients with SSL.
And SSL requires that certificates have an expiration date.
Since we do not want to bother the user with renewing those all the time, we create the certificate with a generous 20 year validity term.
Only problem is that the program we use to create them seems to have a year 2038 problem on 32-bit systems, that wasn't showing until now.

As a workaround for now, set the date backwards before installing:

```
sudo killall ntpd
sudo date --set='2018-1-1'
```

Can correct it after installation again.

rahul-thakoor, 2018-01-25:
The fix worked. Thanks a lot 👍

closed 2018-01-25 by rahul-thakoor