raspberrypi/piserver #13: Error binding to LDAP after running code from Master ( 53c7189 )

mitchellurgero, 2018-02-11:
`Signing certificate...
ldap_sasl_interactive_bind_s: Can't contact LDAP server (-1)`

Pretty weird error, will try a few things and get back to this issue.

maxnet, 2018-02-11:
What Linux distribution are you using?
"Debian Stretch with Raspberry Pi desktop" from raspberrypi.org or something else?

mitchellurgero, 2018-02-11:
Hi - Yes Latest Raspbian Desktop (x86_64?):

Uname: `Linux AIO-001 4.9.0-5-amd64 #1 SMP Debian 4.9.65-3+deb9u2 (2018-01-04) x86_64 GNU/Linux`
lsb: `Distributor ID:	Debian
Description:	Debian GNU/Linux 9.3 (stretch)
Release:	9.3
Codename:	stretch`

mitchellurgero, 2018-02-11:
Since the cert is getting regenerated each time I run, here is a full output of the piserver run function:

`
sudo piserver

** (piserver:3822): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Hit:1 http://security.debian.org stretch/updates InRelease
Ign:2 http://ftp.debian.org/debian stretch InRelease
Hit:3 http://archive.raspberrypi.org/debian stretch InRelease
Hit:4 http://ftp.debian.org/debian stretch-updates InRelease
Hit:5 http://ftp.debian.org/debian stretch Release
Reading package lists...
warning: Unknown type error, skipping line 16
Reading package lists...
Building dependency tree...
Reading state information...
gnutls-bin is already the newest version (3.5.8-5+deb9u3).
nfs-kernel-server is already the newest version (1:1.3.4-2.1).
libnss-ldapd is already the newest version (0.9.7-2).
libpam-ldapd is already the newest version (0.9.7-2).
ntp is already the newest version (1:4.2.8p10+dfsg-3+deb9u1).
ldap-utils is already the newest version (2.4.44+dfsg-5+deb9u1).
openssh-server is already the newest version (1:7.4p1-10+deb9u2).
dnsmasq is already the newest version (2.76-5+rpt1+deb9u1).
The following packages were automatically installed and are no longer required:
  live-boot-doc live-config-doc odhcploc user-setup
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  libsasl2-modules-gssapi-mit | libsasl2-modules-gssapi-heimdal
The following NEW packages will be installed:
  slapd
0 upgraded, 1 newly installed, 0 to remove and 24 not upgraded.
Need to get 0 B/1,498 kB of archives.
After this operation, 17.1 MB of additional disk space will be used.
Preconfiguring packages ...
Selecting previously unselected package slapd.
(Reading database ... 150753 files and directories currently installed.)
Preparing to unpack .../slapd_2.4.44+dfsg-5+deb9u1_i386.deb ...
Unpacking slapd (2.4.44+dfsg-5+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Setting up slapd (2.4.44+dfsg-5+deb9u1) ...
  Moving old database directory to /var/backups:
  - directory unknown... done.
  Creating initial configuration... done.
  Creating LDAP directory... done.
invoke-rc.d: policy-rc.d denied execution of start.
Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Force reconfiguration of existing nslcd
invoke-rc.d: policy-rc.d denied execution of restart.
Generating a 256 bit EC/ECDSA private key...
Generating a self signed certificate...
X.509 Certificate Information:
	Version: 3
	Serial Number (hex): 5a80980029ff832151314dec
	Validity:
		Not Before: Thu Jan 01 00:00:00 UTC 1970
		Not After: Thu Dec 31 23:59:59 UTC 2037
	Subject: CN=piserver
	Subject Public Key Algorithm: EC/ECDSA
	Algorithm Security Level: High (256 bits)
		Curve:	SECP256R1
		X:
			00:dd:1d:d4:03:a7:d3:b5:62:f7:ea:1d:d4:f3:bb:6f
			a2:1a:5f:25:e6:63:4b:78:65:2d:fd:88:89:16:29:b0
			3f
		Y:
			00:82:15:e4:98:a8:ea:b3:39:24:00:15:2a:c2:c5:39
			21:d0:11:04:2b:c0:ad:45:2f:6f:8b:ce:4b:68:c2:32
			f5
	Extensions:
		Basic Constraints (critical):
			Certificate Authority (CA): FALSE
		Key Purpose (not critical):
			TLS WWW Server.
		Key Usage (critical):
			Digital signature.
		Subject Key Identifier (not critical):
			93fb5b7a77177e2b0fe66273f52f1c302c10c9bd
Other Information:
	Public Key ID:
		sha1:93fb5b7a77177e2b0fe66273f52f1c302c10c9bd
		sha256:7a3e73ecf79878854774599cccdd64afc48ac7dc0f9c658dfcc02adb2f1adfad
	Public key's random art:
		+--[SECP256R1]----+
		|       ..+       |
		|        + .      |
		|         . o     |
		|         .E +    |
		|        S  . o   |
		|         o    .o |
		|        .   .+o.o|
		|         . *+o=.=|
		|          =o+oo=*|
		+-----------------+

Signing certificate...
ldap_sasl_interactive_bind_s: Can't contact LDAP server (-1)
`

maxnet, 2018-02-11:
Am unable to reproduce your problem.

Does `sudo systemctl status slapd` show that the LDAP server did start correctly?

mitchellurgero, 2018-02-11:
only lines shown:

```
Feb 11 13:15:31 AIO-001 systemd[1]: Starting LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)...
Feb 11 13:15:31 AIO-001 slapd[3905]: Starting OpenLDAP: slapd.
Feb 11 13:15:31 AIO-001 systemd[1]: Started LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol).
```

mitchellurgero, 2018-02-11:
I would like to note that it does after some time switch to saying something about being booted off network boot (I'll edit this comment if it does this again.)

maxnet, 2018-02-11:
>only lines shown:

- Status is "active (running)", right?

>Feb 11 13:15:31 AIO-001

- Noticed you changed the hostname from the default raspberry to AIO-001
Verify that you also put the new hostname in `/etc/hosts`, and not just in `/etc/hostname`

- You didn't enable anything that can restrict Unix socket communication such as SELINUX, right?

- ``ls -l /var/run/slapd/ldapi`` does show the socket file exists?

- I assume running ``ldapmodify -Y EXTERNAL -H ldapi:///`` manually also gives the error?

maxnet, 2018-02-11:
>Nothing is shown there, dir does not exist

That is strange.
Does `/var/run/slapd` also not exist?

Should have:

```
pi@raspberry:~ $ ls -l /var/run/slapd
total 8
srwxrwxrwx 1 root     root      0 Feb 11 20:14 ldapi
-rw-r--r-- 1 openldap openldap 84 Feb 11 20:14 slapd.args
-rw-r--r-- 1 openldap openldap  5 Feb 11 20:14 slapd.pid
```

``ps aux |grep slapd`` does show it is running, and that it has command line parameters that include ldapi:/// ?
Should look like:

```
pi@raspberry:~ $ ps aux |grep slapd
openldap  1379  0.0  0.4 1088760 4380 ?        Ssl  20:14   0:00 /usr/sbin/slapd -h ldap:/// ldapi:/// -g openldap -u openldap -F /etc/ldap/slapd.d
```

Furthermore, netstat should show it is listening on the Unix socket:

```
pi@raspberry:~ $ sudo netstat -ap |grep slapd
tcp        0      0 0.0.0.0:ldap            0.0.0.0:*               LISTEN      1379/slapd          
tcp6       0      0 [::]:ldap               [::]:*                  LISTEN      1379/slapd          
unix  2      [ ACC ]     STREAM     LISTENING     19632    1379/slapd           /var/run/slapd/ldapi
unix  2      [ ]         DGRAM                    19618    1379/slapd           
unix  3      [ ]         STREAM     CONNECTED     20615    1379/slapd           /var/run/slapd/ldapi
```

mitchellurgero, 2018-02-11:
I just noticed, I can connect on 389 if I do not use sasl 

Did piserver not install slapd with sasl configured?

maxnet, 2018-02-11:
Looking at your output of ``sudo piserver`` again, I notice a couple lines that are normally not there:

```
invoke-rc.d: policy-rc.d denied execution of start.
[...]
invoke-rc.d: policy-rc.d denied execution of restart.
```

Did you set any policy-rc.d?

mitchellurgero, 2018-02-11:
I have not - Like I said, this is a completely stock Raspbian Desktop install.

maxnet, 2018-02-11:
What does ``cat /usr/sbin/policy-rc.d`` give?

mitchellurgero, 2018-02-11:
```
#!/bin/sh
exit 101
```

maxnet, 2018-02-11:
Well, that explains why it does not work.

Wonder where that file comes from though.
``dpkg -S /usr/sbin/policy-rc.d`` shows it didn't come from a package? (should return "no path found matching pattern")

What Debian Stretch ISO did you use to install?
2017-11-16-rpd-x86-stretch.iso or something else?

mitchellurgero, 2018-02-11:
```dpkg-query: no path found matching pattern /usr/sbin/policy-rc.d```

And the ISO from raspbian.org.

maxnet, 2018-02-11:
raspbian.org does not have any ISOs for x86 only for ARM.
What is the file name of the ISO?

Either way, deleting ``/usr/sbin/policy-rc.d`` should solve your problem.
But the file is not supposed to exist at all.
That's why I am wondering where it comes from.

mitchellurgero, 2018-02-11:
Apologies, I meant raspberrypi.org:
https://www.raspberrypi.org/downloads/raspberry-pi-desktop/

That is the iso I am using.

mitchellurgero, 2018-02-11:
Another note: I just fresh installed (and updated) raspbian full arm image on an RPi3 and it gives the same issues, and has that file (Again all stock, nothing manually installed, etc)

mitchellurgero, 2018-02-11:
Still no go though, here is full output, from deleting that file, to executing piserver:
```
pi@AIO-001:~ $ sudo dpkg -S /usr/sbin/policy-rc.d
dpkg-query: no path found matching pattern /usr/sbin/policy-rc.d
pi@AIO-001:~ $ sudo rm -f /usr/sbin/policy-rc.d 
pi@AIO-001:~ $ sudo systemctl stop slapd.service
pi@AIO-001:~ $ sudo apt-get purge slapd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'slapd' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libodbc1 live-boot-doc live-config-doc odhcploc pwgen sharutils user-setup
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 24 not upgraded.
pi@AIO-001:~ $ sudo systemctl status slapd.service
Unit slapd.service could not be found.
pi@AIO-001:~ $ sudo -i
root@AIO-001:~# rm -rf /etc/ldap*
root@AIO-001:~# rm -rf /var/backups/*
root@AIO-001:~# piserver

** (piserver:14454): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
Hit:1 http://security.debian.org stretch/updates InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Ign:3 http://ftp.debian.org/debian stretch InRelease
Hit:4 http://ftp.debian.org/debian stretch-updates InRelease
Hit:5 http://ftp.debian.org/debian stretch Release
Reading package lists...
warning: Unknown type error, skipping line 16
Reading package lists...
Building dependency tree...
Reading state information...
gnutls-bin is already the newest version (3.5.8-5+deb9u3).
nfs-kernel-server is already the newest version (1:1.3.4-2.1).
libnss-ldapd is already the newest version (0.9.7-2).
libpam-ldapd is already the newest version (0.9.7-2).
ntp is already the newest version (1:4.2.8p10+dfsg-3+deb9u1).
openssh-server is already the newest version (1:7.4p1-10+deb9u2).
dnsmasq is already the newest version (2.76-5+rpt1+deb9u1).
The following packages were automatically installed and are no longer required:
  live-boot-doc live-config-doc odhcploc pwgen sharutils user-setup
Use 'apt autoremove' to remove them.
Suggested packages:
  libsasl2-modules-gssapi-mit | libsasl2-modules-gssapi-heimdal
The following NEW packages will be installed:
  ldap-utils slapd
0 upgraded, 2 newly installed, 0 to remove and 24 not upgraded.
Need to get 0 B/1,689 kB of archives.
After this operation, 17.8 MB of additional disk space will be used.
Preconfiguring packages ...
Selecting previously unselected package slapd.
(Reading database ... 150778 files and directories currently installed.)
Preparing to unpack .../slapd_2.4.44+dfsg-5+deb9u1_i386.deb ...
Unpacking slapd (2.4.44+dfsg-5+deb9u1) ...
Selecting previously unselected package ldap-utils.
Preparing to unpack .../ldap-utils_2.4.44+dfsg-5+deb9u1_i386.deb ...
Unpacking ldap-utils (2.4.44+dfsg-5+deb9u1) ...
Setting up ldap-utils (2.4.44+dfsg-5+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Setting up slapd (2.4.44+dfsg-5+deb9u1) ...
  Moving old database directory to /var/backups:
  - directory unknown... done.
  Creating initial configuration... done.
  Creating LDAP directory... done.
Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Force reconfiguration of existing nslcd
[... 46 lines cut ...]
ldap_sasl_interactive_bind_s: Can't contact LDAP server (-1)
root@AIO-001:~# systemctl status slapd.service 
● slapd.service - LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)
   Loaded: loaded (/etc/init.d/slapd; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-02-11 15:00:46 CST; 33s ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/slapd.service

Feb 11 15:00:46 AIO-001 systemd[1]: Starting LSB: OpenLDAP standalone server (Lightweight Director
Feb 11 15:00:46 AIO-001 slapd[14893]: Starting OpenLDAP: slapd.
Feb 11 15:00:46 AIO-001 systemd[1]: Started LSB: OpenLDAP standalone server (Lightweight Directory
lines 1-10/10 (END)...skipping...
● slapd.service - LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)
   Loaded: loaded (/etc/init.d/slapd; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-02-11 15:00:46 CST; 33s ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/slapd.service

Feb 11 15:00:46 AIO-001 systemd[1]: Starting LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)...
Feb 11 15:00:46 AIO-001 slapd[14893]: Starting OpenLDAP: slapd.
Feb 11 15:00:46 AIO-001 systemd[1]: Started LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol).

```

mitchellurgero, 2018-02-11:
Oh that weird error I was trying to describe in a previous comment just came up after a reboot:

```
● slapd.service - LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)
   Loaded: loaded (/etc/init.d/slapd; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-02-11 15:03:50 CST; 3min 3s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 586 ExecStart=/etc/init.d/slapd start (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/slapd.service

Feb 11 15:03:48 AIO-001 systemd[1]: Starting LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)...
Feb 11 15:03:48 AIO-001 slapd[586]: Starting OpenLDAP: slapderror: unexpectedly disconnected from boot status daemon
Feb 11 15:03:50 AIO-001 slapd[586]: .
Feb 11 15:03:50 AIO-001 systemd[1]: Started LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol).
```

maxnet, 2018-02-11:
>Another note: I just fresh installed (and updated) raspbian full arm image on an RPi3 and it gives the 
>same issues, and has that file (Again all stock, nothing manually installed, etc)

What is the exact name of image used?
And what file modification date does the policy file has?

mitchellurgero, 2018-02-11:
policy-rc.d does not exist in image, but image name is `2017-11-29-raspbian-stretch-lite.img`

maxnet, 2018-02-11:
You will have to be a little bit more specific about your problems.
I interpreted "it gives the same issues, and has that file" as that it does have the policy-rc.d file that is not supposed to be there.

Problem is your issues are not reproducible for me.
Works fine with a clean ``2017-11-16-rpd-x86-stretch.iso`` install under Virtualbox.
Could be that you get different updates than me though. Some mirrors are more up-to-date than others.

mitchellurgero, 2018-02-11:
What do you need to know from me?

mitchellurgero, 2018-02-11:
As far as I can tell, both my AOI-001 and Raspberry Pi 3 have pretty much similar rootfs's. Aside from one being x86, and the other being ARM. both updating from the same mirrors in sources.list, both using Raspbian, same version (November image, different dates as seen on the site). 

I wonder if maybe this would be an issue for slapd devs?

maxnet, 2018-02-11:
Have you tried without the updates?

So just with a fresh image, NO upgrades, NO configuration changes (not through raspi-config either).
Just:

```
sudo apt-get update
sudo apt-get install devscripts debhelper cmake libldap2-dev libgtkmm-3.0-dev libarchive-dev libcurl4-openssl-dev intltool gksu git binfmt-support qemu-user-static
git clone --depth 1 https://github.com/raspberrypi/piserver.git
cd piserver
debuild -uc -us
cd ..
sudo dpkg -i *.deb
sudo piserver
```

mitchellurgero, 2018-02-11:
Oh damn, I could certainly try - this (AOI-001) is my main machine though at the moment.

mitchellurgero, 2018-02-11:
shittttt OK will be back in a bit when the image is done reinstalling to HDD.

mitchellurgero, 2018-02-11:
And would you look at that, I just tested on both my desktop (AIO-001) and RPi3:

- Both were reimaged with redownloaded latest image from raspberrypi.org
- one was updated, the other was not.
- built piserver from source on both.
- ran piserver with sudo on both, the one that was updated with apt upgrade FAILED, the other did not. 

So something is changed when an upgrade occurs on the system past November.