raspberrypi/piserver #14: Bridge network adaptors

jonwitts, 2018-02-28:
Hi there,

I am interested in bridging the network adaptors of my PiServer so that my Raspberry Pis can use the internet connection of the WiFi on my PiServer. Is this a configuration that could work with PiServer?

Thanks,
Jon

maxnet, 2018-02-28:
Technically you can.

However in such cases I would personally prefer running a standalone dhcp server on piserver, and NATing Internet traffic to wifi above interface bridging.

Keep in mind that if wifi goes down at some point, and you are using bridging, clients will be unable to renew the IP lease they got from the DHCP server over wifi.
If wifi is down long enough for lease to expire, clients will not just have no Internet, but also lose connection to piserver.

jonwitts, 2018-02-28:
Ah, OK that makes sense. 

I currently have the PiServer running as a stand alone DHCP server. I will have a look at NATing to the WiFi connection. Would I set the gateway for the Pis to the IP address of Eth0 on the PiServer?

maxnet, 2018-02-28:
Yes. The piserver IP would then be the gateway for the clients.
Steps would be something like the following (untested):

1) make sure the computer running piserver has a static IP in a private range not used elsewhere in your network.
E.g. to give it 10.3.14.1 put in /etc/dhcpcd.conf :

```
interface eth0
static ip_address=10.3.14.1/24
````

2) configure piserver to act in standalone mode, on eth0
handing out IP-addresses in the 10.3.14.x range, netmask 255.255.255.0, gateway 10.3.14.1

3) test that network booting the clients works properly.

4) now share the Internet connection through NAT with shell commands among the lines of:

```
echo "1" > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
```

In which wlan0 is the interface that has Internet (could also be something like eth1, if you are running on Virtualbox, and you are exposing wifi to the VM as second virtual NIC) 
If that works properly put the lines in `/etc/rc.local` or another startup script to make permanent.

maxnet, 2018-07-05:
Assuming you managed to get this to work. If not, just reopen.

closed 2018-07-05 by maxnet

OkenKhuman, 2020-10-26:
Hi,
With RPi4 as host I tried `iptables -t nat -A POSTROUTING -o usb0 -j MASQUERADE`
but I end up with
```
iptables v1.8.2 (nf_tables): Chain 'MASQUERADE' does not exist
Try `iptables -h' or 'iptables --help' for more information.
```