raspberrypi/piserver #15: Mount shared directory to all clients

jonwitts, 2018-02-28:
It would be useful to be able to mount a directory from the PiServer to all clients that boot from it. I appreciate that this would be possible by altering the Raspbian image on the PiServer, but would a GUI to achieve this be useful for educators using the PiServer?

Thanks,
Jon

maxnet, 2018-02-28:
Is the desire just to make a folder that the teacher prepared on piserver available on all clients read-only?

Or are you proposing shared folders multiple users should be able to write to?
If the latter, things like user groups and file system permissions would need more thought...

jonwitts, 2018-02-28:
In the first instance the ability for the teacher to share resources to all users read-only would be ace. I could see the desire for read-write access too, but I would not imagine that that use case would be as common...

closed 2018-07-05 by maxnet