raspberrypi/piserver #17: Add option to name operating system during install

jmccoy555, 2018-03-26:
Hi, Would it be possible to add the above to the GUI? 

I have accomplished this by renaming the os dir and adjusting the linking in the .json files post install at it works, but it would be nice to be able to do it during the installation process. 

Use-case, I want different customised versions of Rasbian for different users / pis, but if you already have one installed called 'Rasbian' the GUI does not let you install another as the name will be the same.

maxnet, 2018-03-26:
Currently it is possible for the typical user (that does not customize their image) to upgrade to a later Raspbian version by a simple click on the "upgrade" button, which downloads and replaces the existing image with the latest version.
That would no longer work if renamed though.

jmccoy555, 2018-03-31:
Thanks for the tip! Are updates planned often? It looks like the PiServer Raspbian images are nearly 5 months old and a lot of updates get pulled on the initial update. Ideally a Raspbian 'Essentials' image would be very nice for embedded / industrial applications that don't require all the 'educational, etc., etc......' goodies that come as standard as it's harder to final alternatives in the required tar.xz format. 

A clone button would be very nice indeed, save a lot of downloading and updating, and also act as a sort of image backup option.

maxnet, 2018-03-31:
Will leave the question about when the images will be updated open for someone else to answer. (not my department).

>Ideally a Raspbian 'Essentials' image would be very nice for embedded / industrial applications

Doesn't "Raspbian Lite" fit the bill for that?

>it's harder to final alternatives in the required tar.xz format.

You should be able to convert any Raspbian based image to the right format yourself.
- Create tarballs of the contents of the FAT boot partition, and ext4 root partition.
- Run: /var/lib/piserver/scripts/convert.sh boot.tar.xz root.tar.xz image_suitable_for_piserver.tar.xz

XECDesign, 2018-03-31:
The idea is to release piserver archives with the regular images. There are still some issues to iron out though.

jmccoy555, 2018-03-31:
Sounds good! I have just manually pulled and saved the tar.xz (to save waiting for the download all the time), and when you install from local file via the GUI it names it 'piserver', so I'd suggest an option to name is definitely needed when installing this way. 

Raspbian Lite is lite, but I don't think so bare-bones minimal in comparison to the option in Ubuntu. i.e. what I was thinking is - get me to a log in prompt with a network connection and that's it. I think it's much easier to install what you need with a script than trying to remove what you don't to speed up update / upgrade etc. 

I'll prob give the conversion ago at some point. Thanks.

maxnet, 2018-03-31:
>when you install from local file via the GUI it names it 'piserver',

It currently names it after the tarball you added.
If you call your tarball My_Operating_System.tar.xz it would become "My Operating System"

>Raspbian Lite is lite, but I don't think so bare-bones minimal in comparison to the option in Ubuntu. i.e. 
>what I was thinking is - get me to a log in prompt with a network connection and that's it.

That's pretty much Raspbian Lite (minimal system + ssh + wifi + bluetooth)
Think the only reason it may appear larger than other minimal Linux distributions is that it contains around 100 MB worth of kernel modules for two different CPU architectures (ARMv6 for Pi 0/1 + ARMv7 for Pi 2/3) though.

If you want a real minimal embedded operating system -only supporting the hardware you need it to, and not anything else-, try creating your own with buildroot and a custom kernel instead.
See the gpioexpander and scriptexecutor repositories as examples.

jmccoy555, 2018-03-31:
> It currently names it after the tarball you added.

So obvious... but a nice workaround to name your installs then!

That sounds like a good idea in fact. One use I have is to deploy Squeezelite to a number of players, and I particularly do not want wifi, bluetooth, hdim etc. so that would save disabling all that. 

Good job it's a holiday weekend!!! Thanks.

commit 49f2c666 (2018-08-02): Add 'clone OS' button

Allow user to have multiple (customized) instances of
a single operating system.

Fixes #17
closed 2018-08-02 by maxnet