raspberrypi/piserver #18: Download stalls

bphermansson, 2018-03-28:
I use the latest Github-version. It compiles and runs fine and the download in the "Installation"-section starts. But the speed soon slows down to nothing. I tried many times, it downloads a few megs, then the downloaded counter stops, the speed decrease and hours left just getting higher.

bphermansson, 2018-03-28:
Also: Had the same problem with the preinstalled version on the latest Raspbian.

maxnet, 2018-03-28:
Does `sudo netstat -np` show it is still connected to the webserver?
Wonder if there is something wrong with error reporting or with server.

As a workaround you do can download the file manually from: http://downloads.raspberrypi.org/raspbian/piserver.tar.xz first.
And then select that file in the Piserver GUI.

bphermansson, 2018-03-28:
Thanks for a rapid answer!
What to look for in the netstat output? It's a long list.

maxnet, 2018-03-28:
Can exclude unix socket stuff from the list by adding --inet
`sudo netstat -np --inet`

Am interested if process "piserver" still has a connection to the webserver it attempts to download the image file from.
Or if it has been disconnected, and piserver failed to display an error about it.

bphermansson, 2018-03-28:
I ran wget with the link to the tar.xz file and that halts to. I then tried wget on my desktop computer and that download, but very slowly and with interrupts. Something with the server?

The server seems connected when the download within Piserver stops:

tcp        0      0 192.168.1.196:60958     93.93.130.39:80         ESTABLISHED 14628/piserver

maxnet, 2018-03-28:
>I ran wget with the link to the tar.xz file and that halts to.

Well, then at least it is not a problem in the piserver code itself...

Is it a 3+ you run wget on?
Might be related to: https://github.com/raspberrypi/linux/issues/2449

==

>I then tried wget on my desktop computer and that download, but very slowly and with interrupts. 
>Something with the server?

Did also notice that the speed you get can differ a lot depending on which server you get redirected to.
And that server IP is indeed also currently on the slow side for me:

```
max@lynx:~$ wget -O/dev/null http://93.93.130.39/raspbian/piserver.tar.xz --header "Host: director.downloads.raspberrypi.org"
--2018-03-28 17:32:50--  http://93.93.130.39/raspbian/piserver.tar.xz
Connecting to 93.93.130.39:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1326903248 (1.2G) [application/x-xz]
Saving to: ‘/dev/null’

/dev/null                        16%[========>                                             ] 213.76M   571KB/s    eta 19m 25s
```

Still not slow enough to stall though.

(speed I get from another random server for comparision:

```
max@lynx:~$ wget -O/dev/null http://downloads.raspberrypi.org/raspbian/piserver.tar.xz
--2018-03-28 17:29:19--  http://downloads.raspberrypi.org/raspbian/piserver.tar.xz
Resolving downloads.raspberrypi.org (downloads.raspberrypi.org)... 93.93.128.133, 93.93.130.104, 93.93.130.39, ...
Connecting to downloads.raspberrypi.org (downloads.raspberrypi.org)|93.93.128.133|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://director.downloads.raspberrypi.org/raspbian/piserver.tar.xz [following]
--2018-03-28 17:29:19--  http://director.downloads.raspberrypi.org/raspbian/piserver.tar.xz
Resolving director.downloads.raspberrypi.org (director.downloads.raspberrypi.org)... 93.93.128.230, 93.93.128.211, 93.93.128.133, ...
Connecting to director.downloads.raspberrypi.org (director.downloads.raspberrypi.org)|93.93.128.230|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1326903248 (1.2G) [application/x-xz]
Saving to: ‘/dev/null’

/dev/null                       100%[=====================================================>]   1.24G  20.8MB/s    in 1m 53s  

2018-03-28 17:31:13 (11.2 MB/s) - ‘/dev/null’ saved [1326903248/1326903248]
```
)

bphermansson, 2018-03-28:
I ran it on an old 3. So I guess it's the servers fault. Will try a bit more tomorrow.

maxnet, 2018-07-05:
Assuming to have been a temporary problem and solved by now.

closed 2018-07-05 by maxnet