raspberrypi/piserver #22: Incomplete install instruction: a PATCHED dnsmasq is required

m214089, 2018-04-15:
Hi,
one cannot get piserver get running on a vanilla debian stretch as dnsmasq is too old. It is patched for the raspbian x86 derivative. 
This is missing in the installation instructions ...

Cheerio

m214089, 2018-04-16:
Hi,
ok, fair enough: I just tried, because I got a new small machine which got trouble with the kernel(drm), i386 packages, and some firmware, which I could go around with vanilla debian (amd64 only). Changing to use dhcpcd with appropriate config files is hopefuly not a big problem. I'll have a look. I have to do more changes in network setup anyway to get slurm running
Do you know of more differences?
Thanks a lot,
Luis

blaggacao, 2018-05-01:
I would also be interested in how to get piserver running on default debian/ubuntu (18.04). I want to double up my main workstation to serve a small training pi network and unfortunately, it is not an option to change to raspbian at the moment.

maxnet, 2018-05-02:
>I want to double up my main workstation

Suggest you run software like this in a virtual machine instead.

Your main workstation is not really a place you want to experiment with software that has not been tested on your Linux distribution, and modifies settings that can affect both your network connectivity and the ability to login to the system at all (pam configuration).

m214089, 2018-05-02:
I was backporting dnsmasq 2.79 from butcher to stretch - all the changes in the rpi patch included in the changed version of debian for x86  are included. 
I' ll started having a look into the update of dhcpcd which got patched too and change from isc-dhcp-client to dhcpcd. Hopefully that is sufficient ...

A VM failed miserable because of network issues - at least for me.

maxnet, 2018-05-02:
>I was backporting dnsmasq 2.79 from butcher to stretch

Can also just grab the latest tarball from upstream dnsmasq project.
Includes the Debian packaging stuff.

>A VM failed miserable because of network issues - at least for me.

Virtualbox works fine, as long as you set networking to bridge mode (instead of NAT).

blaggacao, 2018-05-02:
> Virtualbox works fine, as long as you set networking to bridge mode (instead of NAT).

Or any chance to get this software running in a docker container on the long term?

maxnet, 2018-05-02:
>Or any chance to get this software running in a docker container on the long term?

Does docker support running GUI applications without jumping through hoops?
Keep in mind that this software is also meant to be usable by less technical users.

blaggacao, 2018-05-02:
> Does docker support running GUI applications without jumping through hoops?

Hmm, not that I know. It would have to become some kind of Web application, which can be made very user friendly, though... `docker --privileged` would give the necessary access to the host system.

I understand that this is a rather "radical" idea given the current course of development, but if that's a vision worth scrutinizing, then be it said that a usage instruction can go around the following lines:

- Install docker
- run `docker run -d --privileged --restart=always -p 8080:8080 -v /mnt/piserver-data:/var/lib/piserver piserver`
- go to http://localhost:8080

It would be rather easy then to use the server in headless mode and access the management panel remotely.

ZanderBrown, 2018-05-02:
This is another argument for splitting piserver in two (daemon & gui), i would rather we didn't move to a web interface though

jmccoy555, 2018-05-02:
For my use-case a web UI would be great, but either way PiServer is a great tool.

maxnet, 2018-05-02:
>EDIT: TFTP, LDAP, DHCP could be easily federated as docker services either through a docker-compose 
>or directly through a docker socket controller via --privileged.

Not sure if I understand you.
You cannot easily divide this into a multi-container setup, without a lot of work.

Containers work reasonable well for services that are designed from the ground up to be isolated.
E.g. in a typical webhosting setup, one can have a container running a webserver, and a totally independent one running a database server, because the database server provides an easy to use means to connect to it and perform administrative actions like create a new database.

However the DHCP/TFTP server (dnsmasq) we have does not really provide any such administrative communication channel.
If you want a configuration change, -like when you add a new Pi to Piserver-, we need to do that in more primitive ways. Generate new configuration files, and restart the daemon.
Easy to do because our GUI operates in the same space as the daemon software, less easy once you start to isolate things.

blaggacao, 2018-05-03:
I have to admit, I have not personally tried this setup but felt tempted in various occasions. One image that I was looking at is https://hub.docker.com/r/andyshinn/dnsmasq/ with more than 5M pull and top ranked on docker hub it might have some argument on his side (phenomenologically speaking)...

It needs the `NET_ADMIN` capability, indeed.

My naive approach would be something around:
- Bind-mount the images folder
- Bind-mount the persistence folders
- Bind-mount shared config folders (for the piserver controller software)
- Spin up and bridge the dnsmasq container to the host network -> No idea if/how that would work exactly...
- Spin up a LDAP docker service, something around https://hub.docker.com/r/danielguerra/alpine-openldap/
- I would write a `docker-compose.yml` to make those components play nicely together

Then the controller container would just write config changes into the bind-mounted shared folder, persisted on host, and then in order for the peers to pick it up, [cycle them through the docker.socket](https://stackoverflow.com/questions/47748802/restart-container-after-another-container-is-deployed).

If implementing server API is an option, I have been very keen on http://editor.swagger.io/ lately. It helps you define the API first, and then gives you some server stubs in over 20 languages... (see generate server menu).

Just thinking loud, although style might convey it it is not meant to contain any implicit imperative! I wouldn't dare.

m214089, 2018-05-18:
I got back to my installation problem and I got all solved 'easily'. Just added the i386 debian repo with the rasbian desktop, retrieved the source packages fro dnsmaq and dhcpcd5, switched off NetworkManager, removed /etc/resolve.conf maintained by NetworkManager, started dhcpcd and dnsmasq services and it did work without any problem. Installed piserver from the git repo, started it and all went fine. I can provide a step-by-step installation guide in case of interest.

m214089, 2018-05-31:
It took some time (I was very busy lately). But here we go. I do hope that I didn't miss any step. I have a running system up and working and I did it twice following the recipe. However: no warranty.

I would be really glad if updates could  be made available.

Following an installation of a vanilla version of Debian stretch
setting the task selection to desktop with gnome and ssh server the
first task is to get the some additional packages (one needs to remove
the cdrom (USB stick) from sources.list):

```bash
cd /etc/apt
vi sources.list
```

Now get the packages, but update apt before this:

```bash
apt-get update
apt-get upgrade
apt-get install firmware-iwlwifi 
apt-get install firmware-realtek
apt-get install emacs
apt-get install git
apt-get install g++
apt-get install gfortran
apt-get install thunderbird
apt-get install devscripts debhelper cmake git
apt-get install libldap2-dev libgtkmm-3.0-dev libarchive-dev libcurl4-openssl-dev intltool gksu
apt-get install binfmt-support qemu-user-static
```
Next get the piserver sources from github and build a deb-package:

```bash
git clone --depth 1 https://github.com/raspberrypi/piserver.git
cd piserver
debuild -uc -us
```

As we want use the vanilla Debian stretch we need to change a couple
of things related to networking. Therefore we need to add 

```
deb-src http://archive.raspberrypi.org/debian stretch main
```
in /etc/apt/sources.list. Than we need the Raspberry Pi Foundation GPG key:

```bash
wget http://archive.raspberrypi.org/debian/raspberrypi.gpg.key
apt-key add raspberrypi.gpg.key
apt-get update
```

and can create the new packages after updating apt:

```bash
apt-cache showsrc dnsmasq
apt-get source dnsmasq=2.76-5+rpt1+deb9u1
dpkg-source -x dnsmasq_2.76-5+rpt1+deb9u1.dsc 
mk-build-deps -i -r dnsmasq

apt-cache showsrc dhcpcd5
apt-get source --download-only dhcpcd5=1:6.11.5-1+rpt5
dpkg-source -x dhcpcd5_6.11.5-1+rpt5.dsc
mk-build-deps -i -r dhcpcd5

cd dhcpcd5-6.11.5/
debuild -uc -us
cd ..
cd dnsmasq-2.76/
debuild -uc -us
```
Now replace the old network management with the new one. Switch off NetworkManager:

```bash
systemctl stop NetworkManager
systemctl disable NetworkManager
```
Disabling NetworkManager keeps a link of /etc/resolve.conf up, which prevents the later
tools to properly start up. So need to remove this link:

```bash
rm /etc/resolv.conf
```
Install our new newtowrk deb-packages:

```bash
dpkg -i dhcpcd5_6.11.5-1+rpt5_amd64.deb
systemctl enable dhcpcd
systemctl start dhcpcd
dpkg -i dnsmasq_2.76-5+rpt1+deb9u1_all.deb dnsmasq-base_2.76-5+rpt1+deb9u1_amd64.deb dnsmasq-utils_2.76-5+rpt1+deb9u1_amd64.deb
systemctl enable dnsmasq
systemctl start dnsmasq
```
and finally the piserver:

```bash
dpkg -i piserver_0.2_amd64.deb
```

Now follow the standard recommendations for piserver usage.

If we need to enable the ssh daemon on the client in case of the lite
image, one can do the required changes vi piserver.  In the main
piserver interface you get after initial installation: go to the “#
software” tab -> press the console button. This gives you a qemu
powered chroot shell into the installed operating system image, in
which you can make tiny modifications.  If you want to enable SSH
server functionality make sure that you do not only enable SSH, but
also generate SSH host keys.  Raspbian normally does that on first
boot instead, but since the file system is read-only to the clients
that is not possible for them to do.

```bash
ssh-keygen -A -v
systemctl enable ssh
```

maxnet, 2018-08-02:
>I would be really glad if updates could be made available.

Grabbing the patched dnsmasq sources from Raspbian's repository is indeed one way to do it.
Do note that the patch has been upstreamed though, and upstream also has Debian package build scripts.
So may be easier to just download and compile latest official version instead.

Will add some basic instructions for that to the main page.

commit 98d7fe1a (2018-08-02): Mention need for up-to-date or patched dnsmasq

Closes #22
closed 2018-08-02 by maxnet