raspberrypi/piserver #24: Assign static IP

rprr, 2018-04-23:
I have setup piserver to be a standalone dhcp server. How can I set the IPs it assigns to be static based on Mac address?

Thanks.

maxnet, 2018-04-23:
Currently not supported in the GUI.
Not sure if there is much demand for that.

rprr, 2018-04-24:
Thanks. But how can I set it via editing files. Your help is appreciated.

maxnet, 2018-04-24:
Well, in `/etc/dnsmasq.d/piserver` you have lines like:

``
dhcp-host=11:22:33:44:55:66,set:piserver
``

And that would need to become:

``
dhcp-host=11:22:33:44:55:66,set:piserver,1.2.3.4,hostnameyouwant
``

There is a slight problem editing that file though.
Your changes will be overwritten by piserver eventually...

rprr, 2018-04-24:
Thank you. Would there be a way to prevent that?

rprr, 2018-04-24:
Thanks. Maybe I will try the first method.
I suppose the alternative is to have an external DHCP server.

rprr, 2018-04-24:
I was able to get an external router to work serving static IPs.

closed 2018-04-24 by rprr