raspberrypi/piserver #26: piserver tftpboot across subnets

rprr, 2018-04-28:
Is it possible for the piserver to work across different subnets? For example, the piserver computer would be at 192.168.1.10 while the clients would be 192.168.2.11-20 and 192.168.3.11-20.

rprr, 2018-04-28:
Thank you for your reply. By 3+ do you mean, any pi with model 3 or above or is this restricted to the recently released 3b+. 

I will read up on how the ip helper-address command works.

maxnet, 2018-04-28:
3B+
For older Pi you need the SD card with bootcode.bin, if the tftp server is only reachable through using gateway.

Depending on what you are trying to achieve exactly, also other alternatives may be available.
E.g. if your goal is to have Pi at multiple locations perform network booting over the Internet, building a VPN between locations may also work.

rprr, 2018-04-29:
Thanks for the alternative suggestion. I will explore this and other options as well.

closed 2018-04-29 by rprr

MeCJay12, 2019-06-05:
I don't mean to open a closed thread but I have an extension of this question. To boot a client from a PiServer on a different subnet, you need DHCP relay setup on the client's network. What if the DHCP server is on the client's network and a DHCP relay can't be setup in parallel? What DHCP options and settings need to be added?

An example: I have a Pi on my local network 192.168.3.X and it is trying to boot from 192.168.1.138. The 3.X subnet's DHCP server is the default gateway (pfSense). PfSense cannot run a DHCP relay and a DHCP server. What settings would I add to my DHCP server to get it to boot across subnets or do I need to setup a separate system to do DHCP relay in parallel?

maxnet, 2019-06-05:
Using a non-patched DHCP server will only work if it are Pi 3+.
Not for Pi 3 without the +

DHCP response must include the magic string "Raspberry Pi Boot" and next-server IP must point to the Piserver tftp server.

MeCJay12, 2019-06-06:
Am I missing something with this config?
[image]

maxnet, 2019-06-06:
No need to mention bootcode.bin
(But shouldn't hurt to have more options than necessary either)