raspberrypi/piserver #28: Raspbian Stretch 2018-04-18... When OS available in PiServer ?

marc-odp, 2018-05-10:
My Raspbian OS version in PiServer is 2017-11-29. When will the repository be updated ? The upgrade button is disabled in PiServer.
Thanks.

XECDesign, 2018-05-10:
With the current firmware, the pi does not reboot when booted off USB or the network. I'm waiting for that issue to be resolved before pushing out an update.

poleg-on, 2018-07-20:
i have work image of my raspberry. I made many changes for my interest. How can i build piserver os using my image?

maxnet, 2018-07-26:
Closing original issue as 2018-07-05 is in repo.

==

>i have work image of my raspberry. I made many changes for my interest. How can i build piserver os 
>using my image?

On a desktop computer running Linux with a SD card reader attached:

* Put your SD card in reader
* Run `dmesg` and look at last lines to determine which drive it is (e.g. /dev/sdb /dev/sdc etc.)
* Take tarball of partitions.

```
sudo bash
mount /dev/sdX1 /mnt
tar cJf boot.tar.xz -C /mnt .
umount /mnt
mount /dev/sdX2 /mnt
tar cJf root.tar.xz -C /mnt .
umount /mnt
```

* Follow instructions above to convert the tarballs to a format suitable for piserver.

closed 2018-07-26 by maxnet

poleg-on, 2018-07-27:
i created boot.tar.xz and root.tar.xz

after i run converter:
root@debian:/home/poleg# piserver/scripts/convert.sh boot.tar.xz root.tar.xz TT_printserve_for_piserver.tar.xz
Extracting original files
Installing extra software
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "ru_RU.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory
Err http://mirrordirector.raspbian.org jessie InRelease

Err http://mirrordirector.raspbian.org jessie Release.gpg
  Could not resolve 'mirrordirector.raspbian.org'
Err http://archive.raspberrypi.org jessie InRelease

Err http://archive.raspberrypi.org jessie Release.gpg
  Could not resolve 'archive.raspberrypi.org'
Reading package lists... Done
W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/InRelease

W: Failed to fetch http://archive.raspberrypi.org/debian/dists/jessie/InRelease

W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/Release.gpg  Could not resolve 'mirrordirector.raspbian.org'

W: Failed to fetch http://archive.raspberrypi.org/debian/dists/jessie/Release.gpg  Could not resolve 'archive.raspberrypi.org'

W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists...
Building dependency tree...
Reading state information...
The following extra packages will be installed:
  ldap-utils libhx28 libldap-2.4-2 nscd nslcd nslcd-utils
Suggested packages:
  libsasl2-modules-gssapi-mit libsasl2-modules-gssapi-heimdal ncpfs davfs2 xfsprogs tc-utils kstart
The following NEW packages will be installed:
  ldap-utils libhx28 libnss-ldapd libpam-ldapd libpam-mount nscd nslcd nslcd-utils sshfs
The following packages will be upgraded:
  libldap-2.4-2
1 upgraded, 9 newly installed, 0 to remove and 347 not upgraded.
Need to get 1191 kB of archives.
After this operation, 2118 kB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  libldap-2.4-2 libhx28 nslcd libnss-ldapd libpam-ldapd ldap-utils nscd sshfs libpam-mount nslcd-utils
E: There are problems and -y was used without --force-yes
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "ru_RU.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
Looking for files to backup/remove ...
Removing files ...
Removing user `pi' ...
Warning: group `pi' has no more members.
Done.
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "ru_RU.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
The group `pi' does not exist.
Compressing result
Deleting work directory

Complete!
Output is in TT_printserve_for_piserver.tar.xz

maxnet, 2018-07-27:
```
Could not resolve 'archive.raspberrypi.org'
```

Make sure you have working Internet connectivity.

alezzand, 2021-03-08:
> If you don't care about reboots, you do can convert the files manually.
> 
> ```
> wget http://downloads.raspberrypi.org/raspbian_lite/boot.tar.xz
> wget http://downloads.raspberrypi.org/raspbian_lite/root.tar.xz
> /var/lib/piserver/scripts/convert.sh boot.tar.xz root.tar.xz Raspbian_Lite_for_piserver.tar.xz
> ```
> 
> Can then add Raspbian_Lite_for_piserver.tar.xz from local file in the Piserver GUI.
> (remove _lite if you want full version)

Hi Maxnet, 
does this method work with raspios too?
Which one have I to choose (raspios armhf or arm64)?
Thanks...!

maxnet, 2021-03-08:
>Which one have I to choose (raspios armhf or arm64)?

That is up to you.
Think most people are still using armhf.