raspberrypi/piserver #29: Support k8s Helm-Chart deployments

blaggacao, 2018-05-21:
I'm considering to write an [operator for pi-servers](https://github.com/operator-framework), as I have a use case where I want to deploy/take-down pi-servers at disparate remote location connected to a kubernetes cluster to serve all local PIs with centrally managed & standardized images.

However this exciting attempt would require to re implement the pi server as a set of docker-files.

I open this issue to give this idea a voice and have some additional feedback from the core maintainers on how to tackle this...

maxnet, 2018-05-21:
I don't really see the piserver codebase being much use for your purposes.

If you have a need for custom images, a custom user interface (since you want central management rather than a desktop GUI application), a custom runtime environment (docker stuff), and some custom way to distribute images between locations, all which we do not provide out of the box, it will likely be quicker for you to write your own code from scratch, rather than modify the piserver code.

blaggacao, 2018-05-21:
@maxnet Thank you!

I maybe don't even need a user interface just a controller that receives a `ConfigMap` about MAC/ISO mapping for any given deployment and an TFTP interface capable of routing upon this map.

For not implementing any statefulness for now (other than probably a couple of MAC-namespaced configuration files in a special `conf.d`), this might be indeed a good starting point. It is rather a headless use case anyway.

As for the API, just usurping k8s api would be fine, where anyone could build a UI around, if need be.

I should probably next have a look at your "MAC-routing" implementation.

blaggacao, 2018-05-21:
/cc @ychirino

blaggacao, 2018-05-22:
@maxnet 

>Do note that not everyone is that familiar with Docker terminology here.
Does it give my age away if I have no clue what "usurping k8s" implies? blush

Me behaving like an apprentice terminology-spying monster in a way also gives my age away :stuck_out_tongue_winking_eye: 

By this I refer to "ride on the kubernetes api framework for free".

>Problem is that designing a proper API that support two-way communication is a bit of a pain.
Keep in mind that the things we do not always complete instantly, so you cannot get away with a standard request/response API.

CusomResourceDefinition address that problem elegantly by dividing the CRD implementation in a Resource Spec and a Resource Status, where Resource Spec is defined by the user, whereas Status is consumed by the user.

I just copy some lines from my current operator project to give an example:

-- Note the `OdooInstanceSpec` vs `OdooInstanceStatus` both implemented at the highest level of the `OdooInstance`.

```golang
type OdooInstance struct {
	metav1.TypeMeta   `json:",inline"`
	metav1.ObjectMeta `json:"metadata"`
	Spec              OdooInstanceSpec   `json:"spec"`
	Status            OdooInstanceStatus `json:"status,omitempty"`
}

type OdooInstanceSpec struct {
	Image            OdooInstanceSpecImage    `json:image`
	DbSpec           OdooInstanceSpecDbSpec   `json:dbSpec`
	ResourceSpec     OdooInstanceResourceSpec `json:resourceSpec`
	AdminPassword    string                   `json:"adminPassword"`
	ConfigMap        string                   `json:"configMap"`
	SeedingConfigMap string                   `json:"seedingConfigMap"`
	// Replicas         int                      `json:"replicas"`

	// MailServer  bool `json:"mailServer"`
	// OnlyOffice  bool `json:"onlyOffice"`
	// Mattermost  bool `json:"mattermost"`
	// Nuxeo       bool `json:"nuxeo"`
	// BpmnEngine  bool `json:"bpmnEngine"`
	// OpenProject bool `json:"openProject"`
}

type OdooInstanceSpecImage struct {
	Registry string `json:"registry"`
	Name     string `json:"image"`
	Tag      string `json:"tag"`
}

type OdooInstanceSpecDbSpec struct {
	// TODO: Enforce DbQuota compiled by a db Cronjob
	// Using mtDatabase + pg_cron and dbQuota +
	// https://stackoverflow.com/a/37822365
	User        string `json:"user"`
	Password    string `json:"password"`
	MgtDatabase string `json:"mgtDatabase"`
	UserQuota   int    `json:"userQuota"`
}

type OdooInstanceResourceSpec struct {
	Cpu  int `json:"cpu"`
	Ram  int `json:"ram"`
	Disk int `json:"disk"`
}
type OdooInstanceStatus struct {
	DbQuotaUsage  string            `json:"dbQuotaUsage,omitempty"`
	DiskUsage     string            `json:"diskUsage,omitempty"`
	State         OdooInstanceState `json:"state,omitempty"`
	Message       string            `json:"message,omitempty"`
	ImageVersions []string          `json:"imageVersions,omitempty"`
	// Replicas     int               `json:"replicas,omitempty"`
}

// OdooInstanceState ...
type OdooInstanceState string

const (
	// OdooInstanceStateReconciled ...
	OdooInstanceStateReconciled OdooInstanceState = "Reconciled"
	// OdooInstanceStateMigrating ...
	OdooInstanceStateMigrating OdooInstanceState = "Migrating"
)

```

>Then you need to consider if an API is really worth the effort.

The above structs are piped through a code generator framework and throws the readily made API bindings and handler stubs to interact with the specified kubernetes api version.
Essentially, I would argue, that the API comes for free (like in free beer) in this case.

> Keep in mind what the piserver code does -mainly spitting out a dnsmasq configuration file- is not so special.

I kind of smell that, although unfamiliar with dnsmasq and tftp, but that's why I'm pushing it because I feel fear is not warranted.

>It is very easy to reproduce in your own code. So may be less work to just let your own UI code write out the file, then to figure out how to call the API if we offered one.

I would go one step further and just say: provide a ConfigMap as the definition file to the custom resource definition and spin up the piserver container (essential dnsmasq with MAC/ISO mappings) according to this ConfigMap.

> Simple symlink magic.

Stunning.

blaggacao, 2018-05-22:
If I should find a suitable dnsmasq image, I think I've the ~knowledge~ tools to go the first steps towards this idea. It might be easier to understand. May I kindly request to be allowed to name this project `piserver-operator`? Although it would not have (or aim at) feature parity...

blaggacao, 2018-05-22:
OK, then I'd go with `pi-pxe-operator`  :)

> BTW in additon to the symlink MAC/OS mapping that is just for TFTP, we also list the MAC-addresses in the dnsmasq configuration, to tell dnsmasq only to respond to those MAC addresses by (proxy) DHCP.
Just look at the generated config file /etc/dnsmasq.d/piserver to see what I mean.
(Strictly that is not necessary if you are the only server in the network, but it is always better to assume you are not, and only respond to clients you know you are responsible for).

Great! I'll do!

closed 2018-07-05 by maxnet