raspberrypi/piserver #30: Allow more flexibilty for the location of the user file space and the shared directory

gbrickell, 2018-05-29:
Would it be possible for individual paths to be defined in the GUI for the user file space and the shared directory that is now possible (and works really well!). 
I am now working up a PiServer system to use at a school club and am think of getting a reasonably fast laptop with a SSD drive. But a SSD drive may not eventually be big enough to serve the OS images as well as all the shared and user content, so I was going to attach a USB3 drive to the laptop which I'd hoped could be used for the user/shared content storage - but there doesn't seem to be a way at present to change the paths. 
I'm a bit of a newbie when it comes to system detail so I couldn't even think of a workaround.

maxnet, 2018-05-29:
Technically you can mount your SSD drive anywhere you want it to be in the file system.
So nothing prevents you from putting the right lines in `/etc/fstab` to have `/home` and `/var/lib/piserver/os/shared` on partitions of your USB drive.
Do note that in that case your system may fail to boot if the USB drive is not present or breaks though.

Would personally rather get a small normal computer and a normal hard disk than a laptop though.
Can find Intel NUCs for slightly above 100 bucks, excluding memory/storage.

gbrickell, 2018-06-01:
OK - sorry about the delay, only just got back to this - a group of us do this stuff on a voluntary basis - and by way of further explanation the kit we use needs to be portable as we use it in a variety of venues - so a laptop makes sense to us as we don't have to lug around a screen etc for the server. 

I'm doing some initial testing with a very, very old laptop which now has x86 Raspbian, but as I said I'm on a steep learning curve when it comes to the linux system stuff - but I should have thought about mapping the drive with /etc/fstab as that was the suggested way to set up the shared drive - which I did by creating the /var/lib/piserver/os/shared folder and adding the following to the 'served OS' /etc/fstab:
piserver:/var/lib/piserver/os/shared /mnt/shared  nfs  defaults,nolock,nofail,x-gvfs-name=Shared%20folders  0  0

This worked perfectly and a test user logging in at a networked RPi3 could successfully access the shared folder on a read only basis.

But I've hit a problem with mapping the shared folder to a USB drive. I've just used a 32GB USB stick to test the method - so I first of all partitioned and labelled the USB into two primary partitions using 'parted' and as shown below I got the UUIDs:
pi@dellD610_rasp:~ $ sudo blkid /dev/sdb1
/dev/sdb1: LABEL="USER DATA" UUID="0767-5DF0" TYPE="vfat" PARTUUID="74226dfa-01"
pi@dellD610_rasp:~ $ sudo blkid /dev/sdb2
/dev/sdb2: LABEL="SHARED DATA" UUID="05BE-668E" TYPE="vfat" PARTUUID="74226dfa-02"

As a first step I tested the shared folder mapping by editing the main machines /etc/fstab and added the following:
UUID=05BE-668E /var/lib/piserver/os/shared auto nosuid,nodev,nofail  0  0
I may not have all the fstab parameters 'right' but after rebooting and going to the folder on the main machine it correctly accessed the USB stick - BUT when a served RPi3 accesses the shared folder it still 'sees' the unmapped shared folder on the laptop's main hard drive.

Is there a problem with what is essentially double-mapping or should I have used a different set of fstab parameters?? Would appreciate some advice please.

Also having gone through all this then whilst I hope I can get this 'mapping' to work, I'm not sure it is something most folk in the the general 'education space' will want to tackle - so might a development to be able to specify the full paths for these various locations make sense in the long run?

Thanks

maxnet, 2018-06-01:
>I may not have all the fstab parameters 'right' but after rebooting and going to the folder on the main 
>machine it correctly accessed the USB stick - BUT when a served RPi3 accesses the shared folder it 
>still 'sees' the unmapped shared folder on the laptop's main hard drive.

Ah, may be some NFS oddity.
Try if it works better if you edit `/etc/exports` to have explicit entries like this:

```
/var/lib/piserver/os *(ro,no_subtree_check,no_root_squash,fsid=1055,crossmnt)
/var/lib/piserver/os/shared *(ro,no_subtree_check,no_root_squash,fsid=1056,crossmnt)
```

Run `sudo exportfs -ra` to tell the NFS server to reload that file.

> Also having gone through all this then whilst I hope I can get this 'mapping' to work, I'm not sure it is
> something most folk in the the general 'education space' will want to tackle - so might a [...]

Not sure how common the desire to store user data on different USB drives is.

Keep in mind that your request covers a lot more than just creating new home folders at a specific path.
Your extra drives are not automatically available at a path, they need to be mounted through fstab or other means first.

maxnet, 2018-06-02:
And as far as putting `/home` on a separate drive goes, that shouldn't require any special settings (as that does not use NFS, but sshfs).

Nor is it strictly necessary to use gparted, blkid and fstab manually.
Can tell the Debian installer to do that for you instead, like this:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BTW you stated a desire to put just the operating system files on a smallish SSD.
But note that there isn't that much performance to gain from that.
Those files are accessed regularly (by multiple Pi), so will likely remain in the server's memory cache after being read only once.
Linux uses any spare memory not needed by applications automatically as cache. So if your server has modern hardware and has say 2+ GB of memory, that will cover that fine.

gbrickell, 2018-06-02:
OK - thanks for all that and from what you've said then perhaps a laptop with a large conventional hard drive and plenty of memory would be an easier approach.

But just to confirm that your suggested changes to /etc/exports sorted out the 'share' folder access :-)

I will test out moving the /home folder as well and if that is OK I will report back but will then close this thread

Thanks!

gbrickell, 2018-06-05:
Hi - sorry, but I've hit a problem testing the 'mapping' of the /home folder

I added the following to /etc/fstab to also map the /home folder to a partition on the USB stick:
# 1st partition used for USER DATA
UUID=0767-5DF0 /home auto nosuid,nodev,nofail  0  0

... and then rebooted the machine - but it then presented me with a login, just like a client machine although it defaulted to the pi user. But when I used the existing pi password it 'went away' for a second and then just re-presented the login.

To restore the machine and get it working properly again I simply had to remotely SSH into it whilst it was sitting at the login screen and re-edit /etc/fstab to remove the 'mapping'. 

Any suggestions as to what is happening here? Should some of the fstab parameters be different? As mentioned before I am now inclined to just use the conventional set up with a large primary drive - as you recommended!  - but I think it would be useful to get this working so that there is at least a documented approach should it ever become absolutely necessary for someone.

Thanks

Geoff

maxnet, 2018-06-05:
If you want a seperate `/home` mount point I recommend doing so at Debian installation time, instead of later.

If you do it later you will need to transfer your existing `/home/pi` to the new drive.

gbrickell, 2018-06-05:
Sorry - I should have said, but I had already copied across all the /home content to the usb partition before doing the fstab mapping - but I just did it all again in case I missed something. But exactly the same thing is happening.

Any other thoughts?

maxnet, 2018-06-05:
Make sure you preserved file permissions when copying the files. (cp -a)

gbrickell, 2018-06-05:
OK - I hadn't done that previously but I've just repeated it with the -a setting, but still the same thing is happening. :-(

Might be time to give up on this unless you have some more ideas?

maxnet, 2018-06-05:
>To restore the machine and get it working properly again I simply had to remotely SSH into it

And your home directory is there, when you ssh into it?
E.g. doesn't complain about unable to change directory to it, and `ls -al /home/pi` shows your home files are there and owned by pi?

gbrickell, 2018-06-06:
OK - after going round in circles a few times I finally realised that the USB stick was formatted as ntfs and (I didn't know!) that files copied from a linux file system to either of the windows file systems do not preserve the permissions - so even though I used cp -a and this ran OK without any errors it didn't actually preserve the permissions. 

... and yes I really should have checked !

Anyway, I reformatted the USB stick as ext2, copied the existing /home data over and then mapped the /home folder to the USB stick and 'ta da' it all works :-)

Thanks for your patience and support/education on this - and there is now at least a working method to map both the user file space and the shared folder to another drive if anyone runs out of space or needs to do it for some other reason

Thanks again!

maxnet, 2018-06-06:
Good to hear you solved it!

closed 2018-06-06 by maxnet