raspberrypi/piserver #31: installing an up-to-date version of the Arduino IDE

gbrickell, 2018-07-05:
Hi - we are now using PiServer 'in anger' for a School Digital Making Club and have recently started to introduce microcontrollers by using the Pi to program the BBC micro:bit which was easily done by installing mu through the shell of the 'served' Raspbian instance.

But I'm having a bit of difficulty with Arduino, because although its easy to use the standard 'apt-get install' process for the IDE, this installs a very ancient version. Arduino provide downloads of the latest versions (currently 1.8.5) at https://www.arduino.cc/en/Main/Software. Installing the Linux ARM version on a Raspberry Pi with an SD card is very straightforward - you just uncompress the downloaded .tar.xz file and then run the provided install.sh from wherever you did the download/extract. It does give a couple of small install errors about mime files - but it still runs very successfully.

I've tried to mimic this for the PiServer by moving the uncompressed IDE folder to the /tmp folder within the 'served' OS at /var/lib/piserver/os/oursoftwarename/ and then running the install.sh through the served OS shell - and it all goes OK (although it shows the same mime folder errors). But when a client is booted, whilst the Arduino IDE shows up on the Programming menu something is not right as it does not run - and shows an error about the desktop file not being right. 

Looking at the installer it seems to set up everything for the 'current' user - which I guess is 'pi' when the install is done on the PiServer machine - but pi obviously doesn't exist in the client environment - so this might be the cause of the problem?

Do you have any advice/suggestions about how to do this type of install more successfully? My knowledge of this stuff is pretty sketchy (sorry couldn't resist a pun).  I've attached a .zip of the installer in case this helps.

Also I assume I'm right that the install should use the LinuxARM version - even though the PiServer runs on a x86 machine - because the served code will need to run on the Raspberry Pi.

Thanks

Geoff

[arduinoIDEinstaller.zip](https://github.com/raspberrypi/piserver/files/2166123/arduinoIDEinstaller.zip)

maxnet, 2018-07-05:
This is not really a piserver specific issue, as you would have the same problem if you setup multiple users on a single standalone Pi.

>Looking at the installer it seems to set up everything for the 'current' user - which I guess is 'pi' when the 
>install is done on the PiServer machine

The current user would be 'root' if installed in the Piserver chroot shell.
Most installers have the common sense of installing the application at a global location -for every user- if they detect they are being run as root, but apparently that does not apply to the Arduino installer. It will simply install it in root's home directory /root and not for anyone else.

Suggest you ask them if there is any reason why they do it that way...

maxnet, 2018-07-05:
Correction: it does install desktop icons globally if started by root.
Just does not copy it's own application files to any location, and expects that you extracted them to a public location already.

E.g. you can put the files in /usr/local/share/arduino yourself
in chroot shell:

```
cd /usr/local/share
mkdir arduino
cd arduino
wget https://downloads.arduino.cc/arduino-nightly-linuxarm.tar.xz
tar xJvf arduino-nightly-linuxarm.tar.xz
arduino-nightly/install.sh
```

gbrickell, 2018-07-05:
Thanks for looking at this in more detail - that was exactly the problem! Obviously the installer doesn't make this at all clear - but just doing the extract etc in a public location within the served OS, as you suggested - makes it all good :-)

I know this wasn't really a piserver specific issue but as the use of Arduino 'from' the Pi is such a popular digital making thing - this will perhaps be of interest to others that are deploying piserver for groups/clubs/schools

Thanks again

geoff

closed 2018-07-05 by gbrickell