raspberrypi/piserver #33: Docker Image

emkookmer, 2018-08-19:
Is it possible to run PiServer in A Docker image instance?

emkookmer, 2018-08-20:
Hi @maxnet,

I already understond that PiServer will run within a VirtualBox (or any other VM instance for that matter), but my Synology NAS doesn't support creating a real VM instance, but it has Docker support.

closed 2018-08-28 by maxnet

dragon788, 2019-02-06:
It _is_ possible to run GUI applications from within Docker, you can use xpra or VNC within the container or you can share the X server socket from the host into the container. You are correct that it probably wouldn't work correctly on a Synology unless there was another container providing an Xserver.

I actually landed here with the same question, and I'll be looking into a way to do get it running in Docker if possible.

HarvsG, 2019-11-15:
@dragon788 Did you ever have any luck? I would like to have it running in a Docker image on a box running OpenMediaVault.