raspberrypi/piserver #35: Changed piserver ip address client kernel panic

Summer88, 2018-09-05:
Dear sirs, 

I've installed piserver on a debian 9, everything works fine except if I change piserver ip address: clients would go in kernel panic at boot:
 
```
(mount_block_root) from [<80b017e8>]
(mount_root) from [<80b01990>] (prepare_namespace)
....
[end kernel panic -not syncing: VFS: unable to mount root fs on unknown-block(2,)]
```

If  I set the old ip address of the piserver all client cleanly start without errors.
I've only edited  /etc/network/interfaces.d/enp0s3.conf , rebooted, piserver show the correct ip, but fails.
What's missing?

Thanks, BR

maxnet, 2018-09-05:
Yes, we mention that README.md:

---
#### IP update hook

It is expected that the computer that is running piserver:

* is configured to use a static IP-address that never changes.
* -OR- uses dhcpcd as network manager. In which case it will notify piserver automatically if your server's IP ever changes.
* -OR- that you configure your network manager program (e.g. using a ifup/networkd-dispatcher hook script) to execute "piserver --update-ip" every time the system  obtains a new IP lease.
---

If the IP change is a one-time thing just execute "piserver --update-ip" or start GUI and press "save" in the DHCP settings window.

If the computer running piserver is on DHCP and you expect changes more often, you need to add a hook to whatever program is managing your network configuration.
Managing a network connection is one of the most non-standardized things in Linux.
And depending on which Linux distribution, which desktop environment and which packages you installed it can require different instructions.

Summer88, 2018-09-05:
Thank you! Executing `piserver --update-ip` from command line has fixed it.
Sorry I've missed thata part of README.
BR

closed 2018-09-05 by Summer88