raspberrypi/piserver #39: client login now provides a drop down list of userIds

gbrickell, 2018-11-21:
Hi - we have been very successfully using piserver for a School-based club for more than 12 months now and I routinely update the piserver machine as well as the 'served' Raspbian version. In the last few weeks however the client login screen has started to provide a drop down list of all the currently available userIds instead of the user having to type in their individual userId. 

I don't think this is something I have somehow inadvertently done but revealing all the userIds seems somewhat insecure so could this be simply modified by a config somewhere or would it need a code update?

Thanks

Geoff

spl237, 2018-11-21:
When you say "in the last few weeks", can you be any more specific? We pushed an updated build of Raspbian last Friday (16th Nov) and made some additional changes over the weekend - did the problem occur before or after that? Many thanks.

gbrickell, 2018-11-21:
Hi - to be honest I'm not absolutely sure but I think it was there before the 16th Nov - but just to say that I did do another set of full updates just before posting this Issue and the problem is still there

spl237, 2018-11-21:
OK, thanks for checking.

closed 2018-11-21 by gbrickell

spl237, 2018-11-21:
Just FYI - I've found the original problem in one of our packages; we're pushing a fix to our repo in the next day or so, so future updates should be fine. Apologies for the inconvenience.

gbrickell, 2018-11-21:
OK - thanks for letting me know