raspberrypi/piserver #40: timeout/hanging initial connection problems with older RPi3s versus newer B+

gbrickell, 2018-11-21:
Hi - we are experiencing some periodic problems with some client machines failing to connect to the server and just hanging. We are only booting 8 or 9 machines at present and these are not permanently set up but are instead set up for periodic 'club' sessions with an ad hoc network using some gigabit switches and individual ethernet cables.

In all cases the client machines that give a problem are older RPi3s and all the newer RPI3 B+ machines (which obviously have the faster ethernet speed) always boot up OK. 

Not sure how it all actually works or what options there are for the TFTP PXE session but is there a timeout that could be adjusted or might there be some sort of race condition that could occur when multiple clients try to connect?

Ideally it would be good if the piserver could have some optional client connection diagnostics added e.g. an additional screen that tracks and displays the TFTP sessions that start/complete etc. At the moment everything is obviously just a black box.

Thanks

Geoff

maxnet, 2018-11-21:
In addition to having a faster Ethernet speed, the 3B+s also have newer boot code in ROM, which fixes some bugs.

To see if that may be the problem, you do can let the 3Bs use the newer boot code as well, by putting SD cards in them with only bootcode.bin on it ( https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin )

gbrickell, 2018-11-21:
Ok - I've done a quick test with a single old RPi3 at home with bootcode.bin (4.14.79) on an SD card that I had to hand - and it booted OK - but by 'whatsits' law it also boots OK without the SD on the test set up I have at home.

But I will now get a handful of SD cards ready and do a 'proper' test with the full club set up of mixed 3B's and 3B+'s when we next have a session which is this coming Friday. So I will leave the issue open for now and will report back with whatever happens on Friday.

Thanks

gbrickell, 2018-11-25:
Hi - the promising news is that when the 5 older RPI3's were fitted with SD cards with the 4,14.79 version of bootcode.bin all 8 RPi3's booted OK at the last club session - so this may have fixed the problem. But because the problem was a bit intermittent I will leave this issue item open for a while until we've had another couple of club sessions and if they all go OK I will report accordingly and then close this issue.

Thanks for the advice!

gbrickell, 2018-12-09:
Hi - just to say that adding the bootcode.bin to a SD card for all the 'old' Raspberry Pi's has solved this problem with consistent booting now happening for all the Pi's in the PiServer network over the last few club sessions.

I guess this should be documented somewhere since others must be experiencing the same issue. Let me know if I can help with that in some way - not really sure where the PiServer documentation 'lives'.

Thanks again

closed 2018-12-09 by gbrickell