raspberrypi/piserver #42: Extremely Slow

ferrabacus, 2018-12-06:
I am an educator using PiServer and am running a VM of PiDesktop on an iMac Pro with plenty of allocated resources to the VM. This is connected to a network switch that is rated for 1gbps for each port. Each of the 30 raspberry pis are connected to the switch, the iMac is connected to the switch, and the switch is connected to the internet. All with cat6 cable. 

Everything is working great, but I can't help but notice that each raspberry pi runs significantly slower. I understand that booting will obviously take longer. I'm fine with that, but even browsing the internet is significantly slow. 

I thought it might be the switch, but I connected 30 raspberry pis to the same switch running raspbian off an SD card and they were blazingly fast. 

I need that speed, but I need the username control and locale file system so students can login on any raspberry pi. 

Is there anyway I can optimize the speed? or other options to maintain username control?

maxnet, 2018-12-06:
Is speed any better if you browse the Internet in an incognito window?
If so, we could look into disabling the browser's disk caching.

maxnet, 2018-12-12:
That pretty much everything stalls and only mouse cursor still works, do is a symptom you get when connection with the server is lost (easy to reproduce by unplugging network cable). And could also occur if there is something wrong with network, or buggy virtual machine software causing excessive packet loss.
It should recover in +/- 10 seconds the moment connectivity is restored to an acceptable level though.
Not sure why that does not seem to be occurring.

maxnet, 2019-02-07:
It is possible to disable caching.
That do would reduce network traffic, but may only masquerade the only problem.
To do so, in Piserver, software tab, open a (chroot) shell, enter `leafpad /etc/chromium/default` and add the following line:

```
CHROMIUM_FLAGS="--disk-cache-dir=/dev/null --disk-cache-size=1"
```

If it is a managed Ethernet switch, you may also try if downgrading all ports to 100 mbit speed except the one to your server may work better.
While that may sound strange, I have the impression some Ethernet switches may be handling PAUSE frames -which the USB NIC on the 3+ use- in an unfortunate way -broadcasting them to every port- causing total speed of EVERY port including the one to the server to be limited to 200-300 mbit otherwise.

Other than that all kinds of network specific problems (e.g. badly patched cables) can cause this.
Hard to diagnose remotely.

ferrabacus, 2019-02-28:
I do have a managed switch, and this is the configuration for all of them. 

[image]

g47 is the connection to the internet and g48 is the connection to the iMac running Raspbian. 

I do have the Raspberry Pi 3B+ so 1000Mbps is supposed to be supported, but should I change all of those to 100 Full Duplex or half Duplex? 

[image]

maxnet, 2019-02-28:
100 Full Duplex
(and for testing purposes only. if it does not provide any improvement, change it back)

ferrabacus, 2019-02-28:
For port g48, which is the internet link, as well?

maxnet, 2019-02-28:
Not necessary for g47 (link to the Internet, which is already 100 mbit by default)
Nor for your iMac at g48, keep that at 1 gbit.

ferrabacus, 2019-02-28:
[image]

None of the Raspberry Pis are creating stable connections now. This is one of the errors.

ferrabacus, 2019-02-28:
I disabled STP and set it to auto. Then none of the RPi's connected. 

I re-enabled STP. None connected. 

I reset ports to be auto and the connection is re-established. (Figured this out 30 seconds before students rolled in :] )

Later today, I'll try to disable just STP and see what happens. 

Thank you.

ferrabacus, 2019-04-04:
**It is fixed now....**

I was using virtual box and used all the recommended settings, but that added layer must have been too much. 

I installed Raspbian Desktop for Mac and now their user experience is blazing fast. Maybe my VM settings were off, but a dedicated machine seems to be the way to go!

closed 2019-04-04 by ferrabacus