raspberrypi/piserver #47: Improved user management request

gbrickell, 2018-12-21:
Hi - as our use of PiServer in both the local School and at ad hoc local workshops has steadily increased the 'population' of users has expanded. But the users are not a consistent/stable population so it would be useful to have a couple of additional data fields that are attributed to each user so that we can keep better track of the user population and then perhaps off-load/archive data as users 'disappear'.

The things that seem most useful to us at present, that would ideally be shown on the user admin screen are:
- a timestamp for their last login
- a simple text field where we could add add some info about the user

Then could something be added to easily off-load/archive an individual user - with the option to bring them back if needed? 

Does this all seem like a useful and reasonably easy to implement new set of features for the system?

Thanks

maxnet, 2018-12-21:
Do those users only have piserver accounts, or are they also in any other school system?

Wonder how much demand there is for proper user management in piserver, as I expect most larger school already have other solutions for user management.

gbrickell, 2018-12-22:
Not sure if our usage model is particularly unique but we use a laptop for our PiServer as we run a School club as well as using additional kit to run ad hoc workshops e.g. at Youth Clubs/Scout groups etc., where there can also be overlap in the users e.g. the same user at both the School and the Youth Club.

This means we can't use the School's management system - which I assume exists but we are not really at a stage where we can consider any sort of integration with the School's systems anyway.

I can see that there is potentially a slippery slope towards more and more user management functions but I'm not really suggesting that - just hoping that what we've suggested might be considered.

Thanks

marc-odp, 2019-01-07:
We are a school too. We use a Virtual Machine on a Synology DS for PiServer. We have approximately 30 Pis (both Pi2 and Pi3) but many many users. At the moment we use same access account for everybody. I can confirm that Users handling is a challenge. Any suggestion would be welcome.

marc-odp, 2019-01-07:
Could we use for example a python script to synchronize PiServer's users with our School Management System (mysql db) ?

maxnet, 2019-01-07:
>At the moment we use same access account for everybody.

Do you also use the same account for everyone on Windows computers?
Note that piserver can also authenticate users against external LDAP/Active directory servers.

>Could we use for example a python script to synchronize PiServer's users with our School Management 
>System (mysql db) ?

Technically yes.
However that's a lot of work.

maxnet, 2019-01-07:
Ah, just figured out you are an NGO equipping class rooms with Pi-tops, and therefore may not have any Windows computers unlike the average school...

If the school management system you use is not a commercial one (which typically already has support to authenticate Windows computers, which we could use as well) but something home made, your options are:

- Manually exporting a list of usernames from your MySQL database to .csv file using phpmyadmin or similar.

Can then import it in the Piserver GUI.

- Telling piserver's LDAP server to consult your MySQL server directly.

Requires quite a bit of knowledge of SQL queries and LDAP internals.
See "man slapd-sql" and https://github.com/openldap/openldap/tree/master/servers/slapd/back-sql/rdbms_depend/mysql for information on how to setup mappings.

- Modify your school management system software to talk to Piserver's LDAP server and insert users there.

Password that gives write access to the LDAP server can be find in /lib/piserver/settings.json username is: "cn=admin,dc=raspberrypi,dc=local"

You will need to add entries to LDAP that look like this:

```
dn=cn=<username>,dc=raspberrypi,dc=local
uid=<username>
cn=<username>
sn=<username>
objectClass=inetOrgPerson
objectClass=posixAccount
objectClass=shadowAccount
uidNumber=<user number>
gidNumber=100
homeDirectory=/home/<username>
loginShell=/bin/bash
userPassword=<hashed password>
```

"user number" should be in range 10000 ... 4294967295

gbrickell, 2019-03-14:
Hi - the responses to my original request seem to have morphed into a LDAP integration discussion which is not something that we are currently in a position to consider.

So I'm wondering whether my original suggestion of some very limited additions to the User management area (last login timestamp + a simple description field) plus some advice on how to off-load/archive individual users and their file space from the system - might still be considered?

Thanks

maxnet, 2019-03-14:
Problem with extra user management features is that they may take longer to implement than you may think.
E.g. last login time is currently not recorded (at least not in a way in which the date can be retrieved without spitting through logs), so more work than just displaying a value in the GUI...
Would need to adjust the LDAP scheme and install plugins (e.g. lastbind) to record such things to start with.

closed 2019-06-07 by maxnet

maxnet, 2019-06-07:
[image]

Added some minimal support to display last login date, if that information is available in the LDAP database.

If you want to play with it, you will need to compile and install latest source from git.

Last login recording is then enabled for new installations by default.
If you want to do so for existing LDAP installation, you can use the following command to enable it:

```
sudo ldapmodify -Y EXTERNAL -H ldapi:///
dn: cn=module{0},cn=config
changetype: modify
add: olcModuleload
olcModuleLoad: lastbind.la

dn: olcOverlay=lastbind,olcDatabase={1}mdb,cn=config
changetype: add
objectClass: olcLastBindConfig
objectClass: olcOverlayConfig
objectClass: top
olcOverlay: lastbind
olcLastBindPrecision: 60

<press ctrl+d>
```

commit 786914a3 (2019-06-07): Add support for lastlogin tracking, description

Adds basic support for:
- last login date
- freeform user description

Fixes #47