raspberrypi/piserver #51: PiServer Setting up a centralized Kiosk Server - Multiple Pis each with different URLs

gamertoons, 2019-02-11:
Here is my issue. 

We run a company where we will have multiple Pis out in the field (possibly up to 200)

I would like to design the Pis so they launch basically in Kiosk mode.

This means that each Pi, upon boot, would go to their respective URL.

I want these Pis to be centrally managed because if the URL changes or the OS needs updating or even fixing I can do these from a single location. Since these Pis would be out in a nationwide environment, I would like to prevent 200 of them from shipping back to me so remote access seems the best choice.

Can I do something like this with PiServer in its current implementation?

maxnet, 2019-02-11:
If "out in the field" means remotely over the Internet, than it is outside the scope of Piserver.

PXE network booting generally does not work over the Internet.
(Without VPN constructs, DHCP relaying or other advanced network configurations, which you do not mention having in place)

gamertoons, 2019-02-11:
Hi, I was just saying that our network spans multiple states, but they are all on the same forest network. Communication wont be an issue.

maxnet, 2019-02-11:
>I was just saying that our network spans multiple states

Then you will have latency, which is not nice when using NFS to access the root file system.
Wrong tool for the job.

gamertoons, 2019-02-11:
It's just a browser kiosk. Pulling the initial OS upon first boot wont be an issue. I was more or less asking if it was possible to have multiple of the same OSs (one for each mac address)

maxnet, 2019-02-11:
It is certainly possible to have multiple images of the same OS.

Just be aware that file system access does not only occur on boot, but quite a lot after that as well.

Does your organization have a NAS that is located in another state?
As a quick test, try installing Berryboot on SD card, and tell it to use your NAS for storage.
https://www.berryterminal.com/doku.php/storing_your_files_on_a_synology_nas_using_iscsi
If that is fast enough for your purposes, using piserver may be a possibility, but if that already has too much latency forget about it, as nfs can be 5x slower than iscsi over latent links.

gamertoons, 2019-02-11:
Thank you for the explanation!

bbinet, 2019-02-12:
Balena-os is also a interesting project and the balenaDash example may fit your needs:
https://github.com/balena-io-projects/balena-dash
https://www.balena.io/blog/make-a-web-frame-with-raspberry-pi-in-30-minutes/

closed 2019-06-06 by maxnet