raspberrypi/piserver #52: /etc/ld.so.preload ERROR messages

gbrickell, 2019-03-14:
Hi - following some recent updates to our 'served' OS we have started to see the following ERROR message when a client first boots and it is also shown when the Thonny IDE is first started. 

ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored

Obviously not fatal - but quite distracting, especially for students. 

Some Googled posts suggest that it can be simply fixed by removing the line from /etc/ld.so.preload - but perhaps it is just a permission thing in the PiServer environment? 

What is your advice/suggested fix

Thanks

geoff

maxnet, 2019-04-02:
Related to: https://www.raspberrypi.org/forums/viewtopic.php?t=235594

gbrickell, 2019-04-25:
Thanks for the 'Related to:' info - and now that I've updated my served OS a few times the ERROR messages have stopped - so I'm guessing the underlying issue was dealt with through one of these updates. I will close this item now.

cheers

Geoff

closed 2019-04-25 by gbrickell