raspberrypi/piserver #53: pi4j not run: "UnsatisfiedLinkError: /tmp/libpi4j.so"

kisChang, 2019-03-21:
I developed a student program using Pi4j and Java, but it can't run on piSever, and I get an error message:
"UnsatisfiedLinkError: /tmp/libpi4j2288920775624554689.so: failed to map segment from shared object"

I found the reason for the error after searching, because /tmp mounts options containing 'noexec', I deleted it and the program works fine.
For what reason added this 'noexec'? Can it be removed if it is not necessary?

maxnet, 2019-04-02:
It is standard security practice on server and multi-user systems to make /tmp noexec, as it is the number one location used to install malware, and very few legitimate applications have a need to run executables there.

If security is not so important for your use-case, you can remove it.

kisChang, 2019-04-08:
@maxnet ok, I got it, Thanks!

closed 2019-04-08 by kisChang