raspberrypi/piserver #56: Can I add users to other groups when user login?

chengjianwen, 2019-04-17:
I installed raspbian_lite os and want to user to use frame buffer functions. Tt seems they should be in video groups.
I don't known how to add them to video group when user login.
Thanks.

maxnet, 2019-04-17:
In the piserver GUI: go to tab "software" -> select operating system you want to modify and press "shell" button.

leafpad /etc/security/group.conf

There is currently a line there that specifies which extra groups the user will be made member of if he logs in to the lightdm GUI.

```
lightdm;*;*;Al0000-2400;dialout, audio, video, spi, i2c, gpio, plugdev, input
```

You can make a similar line for console logins:

```
login;*;*;Al0000-2400;dialout, audio, video, spi, i2c, gpio, plugdev, input
```

Do be aware that this should only be done if you are operating in an environment in which security is not a concern.
Be aware that the user could switch to a different tty (ctrl+alt+f2), start a mischievous program there, switch back to tty1, and walk away.

When the next user logs in, the program could keylog the other user by abusing the privilege of being member of the input group.
More complicated attacks could do similar things through the video group, as it allows talking to the GPU and the GPU has access to all memory.

That's why by default only extra privileges are granted if the users logs in to lightdm.

chengjianwen, 2019-04-17:
good ways to do that, I'm very thanks for this.

closed 2019-04-17 by chengjianwen