raspberrypi/piserver #58: how to support self service password change

chengjianwen, 2019-04-19:
When the user login, the user should can change their password.
I installed self-service-password and it's config file seems support LDAP.
https://github.com/ltb-project/self-service-password/tree/master/conf
But I don't known how to config that.
or any other solution for that.

maxnet, 2019-04-19:
Note that the user can change it's own password already by running "passwd" on the shell.
And there is also an option to force change password on first login when adding users in piserver.

However if you want to be able to let users reset their password if they forget it through some web portal like that you will need to experiment with it yourself, as we have not tried using such solutions.

Login details needed should be (untested):

Username: cn=admin,dc=raspberrypi,dc=local
Admin password is chosen randomly on piserver installation and can be found in the file: /var/lib/piserver/settings.json 

Base DN: dc=raspberrypi,dc=local
LDAP username attribute: cn

chengjianwen, 2019-04-19:
I tried it and it worked.
in config.inc.php, I filled in the values mentioned by you.
[code]
# LDAP
$ldap_url = "ldap:///";
# $ldap_starttls = true;
$ldap_binddn = "cn=admin,dc=raspberrypi,dc=local";
$ldap_bindpw = "Ddld949TLZlsbdvmVuAhg24WhwXhftKH";
$ldap_base = "dc=raspberrypi,dc=local";
$ldap_login_attribute = "cn";
# $ldap_fullname_attribute = "cn";
# $ldap_filter = "(&(objectClass=person)($ldap_login_attribute={login}))";
[/code]
BTW, I can use passwd also. 
Thanks.

closed 2019-04-19 by chengjianwen