raspberrypi/piserver #59: remote monitor

chengjianwen, 2019-04-20:
Maybe the team of piserver have think about about how to support remote monitor or even remote control functions for piserver. or it can be immplement through install 3rd software to resolve it, for example epoptes.
For my sitiuation, I installed Raspbian and Raspbian lite in piserver. But it seems Epoptes-client only worked on X, and maybe I can hacked some of it's scripts (for example screenshot) to suit for non X environment.
I known there are many methods to get screen shot for Pi. Will the piserver support some features about that? or I should focus on Epoptes or alternative software for that. I think I'll have to hack some codes to do that if I use epoptes.

maxnet, 2019-04-20:
Not sure if I understand the purpose.
People that are using command line Linux are typically mature enough that they do not need that extensive monitoring and assistance, which is often what software like that is used for.

If it is just about running commands remotely on Pi, you could use ssh. (generate host keys first, before enabling sshd)

chengjianwen, 2019-04-20:
Thanks. mabye it is a subject in future.

closed 2019-04-20 by chengjianwen

marc-odp, 2019-08-02:
Well... I can understand the need. This is more about monitoring what students are doing than giving support. Typically, in large lab, the teacher can see on his screen what the students are doing, it helps giving support when needed but also to react if a student is watching videos on Youtube instead of doing his/her work...
I know about Epoptes but I have no idea how it could be used with Piserver. It is working with Pinet though.

maxnet, 2019-08-02:
>but also to react if a student is watching videos on Youtube instead of doing his/her work...

Watching Youtube on Raspbian Lite which is console only?

The issue starter already knew about Epoptes, but expressed concern that it only works under X...

marc-odp, 2019-08-02:
"For my situation, I installed Raspbian and Raspbian Lite in piserver."
Sorry, I didn't realise that his problem was only about Raspbian Lite because X was not there.
Can @chengjianwen confirm that Epoptes works with Rapbian (Full) ?