raspberrypi/piserver #61: Raspberry not showing on network

lgerster, 2019-05-07:
I am trying to network boot some raspberry pis using PiServer. I am using my own Raspbian Stretch Lite disk image with preinstalled software, which I copied from a working SD using the instructions in https://github.com/raspberrypi/piserver/issues/55.
Running from an sd card, the raspberry boots normally and shows up on the network. With PiServer, I see the raspberry in the list, I can select the software, but the raspberry never shows up on the network.
What is going wrong? How can I start debugging what is going on?

maxnet, 2019-05-07:
- Did you try if the standard Piserver images work first?
- Define "did not show up on the network". How did you test that?
- Did you connect a normal HDMI monitor to the Pi, and does it show it is booting (rainbow screen, console, etc.) or does it stay blank?

lgerster, 2019-05-09:
The standard raspian did not show up either. 
I used nmap to check the local network, as well as looking on the browser interface of the router.
The monitor stays blank.

maxnet, 2019-05-09:
- Do "tail -f /var/log/syslog" on the piserver server while you attempt to network boot to see if it generates any messages.
Should see some messages about DHCP requests followed by the Pi retrieving files by tftp.
- Is it a Pi 3+ or an older Pi 3 without the + ? If the latter, you can also try if it works better by putting a FAT formatted SD card in it with only bootcode.bin (and no other files) on it. Fixes some issues in the network boot code.
- You do are using a simple standard Ethernet network switch like most people have at home? Some of the more fancier managed Ethernet switches have settings that can interfere with network booting (STP, DHCP snooping, ACLs, etc.). (Although in most cases you would not be able to add the Pi to Piserver in the first place, as it fails to detect it, in those cases).

lgerster, 2019-05-10:
- I can't see any messages on the machine running piserver, it is an Ubuntu installation. Do I have to start any service but piserver, or do anything special to launch piserver?
- It is an older Pi 3,  I added the bootcode.bin SD card, with no difference
- It is a netgear genie switch

maxnet, 2019-05-10:
>I can't see any messages on the machine running piserver, it is an Ubuntu installation. Do I have to start any service but piserver, or do anything special to launch piserver?

Your unspecified version of Ubuntu may come with a version of dnsmasq that is too old.
Did you follow the instructions called "Extra dependencies if NOT running Raspbian or "Debian Stretch with Raspberry Pi desktop"" on the main github page?

lgerster, 2019-05-10:
I do have the updated dnsmasq installed, but it seems to fail to start up, giving me the message:
systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
sqipeval dnsmasq[26154]: dnsmasq: cannot set --bind-interfaces and --bind-dynamic

maxnet, 2019-05-10:
What do your /etc/dnsmasq.conf and /etc/dnsmasq.conf.d/* look like?

We only set "bind-dynamic", so am guessing "bind-interfaces" was set by something else on your system that also wants dnsmasq, or some older configuration file you once used...
Existing configuration will not be overwritten, when installing new dnsmasq version.

lgerster, 2019-05-10:
It looks like network-manager used bind-interface, disabling /etc/dnsmasq.d/networkmanager lets me start dnsmasq and boot the raspberries.
Now I just have to figure out how to launch my scripts on boot, but thank you very much for your help!

closed 2019-05-10 by lgerster