raspberrypi/piserver #64: Different shared folders per OS

RosemanJ, 2019-06-03:
Would it be possible to create shared folders that can be assigned per OS? Thinking a 1-to-many relationship...

For example:
Create folders "A", "B", "C", and "D"

Raspbian Full mounts "A", "B", and "C"
Raspbian Lite mounts "D"
A cloned-customized OS mounts "B" and "C"

maxnet, 2019-06-03:
You can do so manually.

As in create some folder under /var/lib/piserver/os (that is easiest, as that is already exported by NFS) on the server: 

```
sudo mkdir /var/lib/piserver/os/shared_A
```

In the Piserver GUI, go to tab "software", select the "operating system" press the "console" button.

Create a mount point where you want the shared folder to appear in the file system, and edit /etc/fstab :

```
mkdir /mnt/shared_A
leafpad /etc/fstab
```

Add an entry like:

```
piserver:/var/lib/piserver/os/shared_A   /mnt/shared_A      nfs     defaults,nolock,nofail,x-gvfs-show,x-gvfs-name=Shared%20folders%20Alpha    0       0
```

RosemanJ, 2019-06-03:
Thanks for the quick reply and I appreciate the manual way of doing this... I guess I was just wondering if it's worth the effort to add such functionality to the GUI.

Two "issues" I see are:

1) "shared_A" (obviously) doesn't show up in the GUI and thus needs to be documented and communicated so coworkers know of its existence (not that that's a bad thing!).

2) if one doesn't want shared folders across OSes at all (class "A", using OS "A" is doing object recognition and has an objectRecognitionScripts folder with appropriate examples, whereas class "B", using OS "B" is a "learn to code Python by writing games" and has a pythonGamesScripts folder filled with stubbed game examples), neither folder can be created via the GUI (and correspondingly nothing at all will show up under the Folders tab).  (horrible examples, I know. hahaha)

maxnet, 2019-06-03:
>shared_A" (obviously) doesn't show up in the GUI

Maybe not in the piserver GUI, but it will be on the clients though.
The "x-gvfs-name=Shared%20folders%20Alpha" in my example fstab entry, will give it a friendly name of "Shared Folders Alpha" in open/save dialogs in the GUI.

RosemanJ, 2019-06-03:
> The "x-gvfs-name=Shared%20folders%20Alpha" in my example fstab entry, will give it a friendly name of "Shared Folders Alpha" in open/save dialogs in the GUI.

I saw that, thanks!!