raspberrypi/piserver #68: Raspbian Buster availability?

Lukanite, 2019-06-28:
Hey, just got a Pi 4 today and wanted to migrate my boot image up to Buster so that it can support Pi 4s - I see that the rootfs and bootfs tar.xz's seem to be updated on the website, but Piserver still shows the Version as 2019-04-08.

Will this be updated to Buster or should I just perform the conversion using the script?

Thanks for all the hard work, by the way!

maxnet, 2019-06-28:
Pi 4 does not support network booting yet.

XECDesign, 2019-06-28:
There will be an update coming a little later. Probably next week.

Lukanite, 2019-07-08:
Ah, okay. Currently I'm booting my Pis with the boot partition on the SD card only, nfs rooting to Piserver, and then checking if the boot partition should have any files updated on it, which I think would let me get around this, but I can wait until the firmware's a bit more ready.

Thanks for the info - looking forward to it!

ferrabacus, 2019-07-09:
> Pi 4 does not support network booting yet.

Will it be updated for us to use buster with 3B+? 

and is there a roadmap for network booting pi4? Because I may have just ordered 40 pi4 to use with my current piserver setup.

XECDesign, 2019-07-09:
There will be a buster piserver image when we've had a chance to test it.

gollygosh, 2019-07-10:
Just for information I did try to install my own custom buster image from .tar.xz's and it failed on:

qemu: Unsupported syscall: 383

I assume this is some kind of issue like a glibc mismatch in the chroot image.

maxnet, 2019-07-11:
>qemu: Unsupported syscall: 383

If your Piserver computer is currently running Debian Stretch, you may need to borrow a newer qemu version from a more recent Debian version.

http://ftp.de.debian.org/debian/pool/main/b/binfmt-support/binfmt-support_2.2.0-2_i386.deb
http://ftp.de.debian.org/debian/pool/main/q/qemu/qemu-user-static_3.1+dfsg-8~deb10u1_i386.deb
(install in that order, binfmt first)

maxnet, 2019-07-12:
Even with proper qemu Buster's kernel currently hangs shortly after boot on my Pi 3+ though.
So don't bother running convert.sh yourself right now.

Created an issue: https://github.com/raspberrypi/linux/issues/3067