raspberrypi/piserver #70: RPi documentation - despite intended audience is still not gender neutral!

arclite7, 2019-07-02:
I ask politely for a collective responsibility to extend itself to assure all RPi documentation is maintained with exemplary gender neutrality. There is every reason to applaud the continued results of international collaboration from both ends of the RPi business model. From all ages of contributors and with equal significance. 

There has never been such an opportunity for British détente than by the development of a completely open, accessible and equal presentation of a combined hardware, firmware and software technology, a.k.a. a SBC that has revolutionised educational opportunities, enabled creative experiences whilst it is subsidised and supported by a global social media network and a global 24hr marketplace, viewable through the same technology.

Since the loss of gender neutrality is evident here on Github it is assumed a complaint or request for change is submitted directly here.

maxnet, 2019-07-02:
No documentation maintained by this project.

closed 2019-07-02 by maxnet

arclite7, 2019-07-03:
You are wrong MAXNET: please to review your decision with following documentation reference:

   **### A minimal support for external LDAP/AD servers**

- Adds page to installation wizard to ask wheter user likes to
  create users with Piserver or use an external auth server.
- If user selects he wants to use external server Piserver will
  do a quick attempt to auto-detect the location of said server.

...as this was being responded to. If you are not willing to adjudicate reasonably or fairly on matters of GENDER NEUTRALITY it is reasonable for this matter to be escalated to include an equal number of responsible personnel with a "different gender" to that of your own as per European equality guidelines. 

**Women in decision-making: why it matters | European Institute ...**
[Search domain eige.europa.eu/news/women-decision-making-why-it-matters] https://eige.europa.eu/news/women-decision-making-why-it-matters
With the domain of power holding the lowest score in our Gender Equality Index, women remain under-represented in the kinds of decision-making positions that shape politics, economics and society. Yet gender equality is a founding value of the EU and improving it could lead to an increase in GDP of up to €3.15 trillion by 2050.

I certainly do not regard the issue of gender neutrality as trivial that it can be deprecated without reasonable response to the central issue! Without a single request for clarification?  The RPi business model approaches the most vulnerable people worldwide and all prospective references from that model must take responsibility for consolidating a gender neutral position in all aspects of their presentation to the RPi audience. Unless you would prefer to go head to head with a single gender supremacist perspective based upon a model of pre-existing social privilege? Please do explain your position for the record if you will? If you choose not, please be informed of my intention to make a formal complaint of your action since as I am a newcomer to Github you have failed, so far,  to make any aspect of this "negotiation" clear or informative. Gender neutrality may in fact be new to yourself. The RPi model is itself an attempt to make immediate access to cross referencing of factual information second nature to the forthcoming generations such that international collaboration will be the next revolution. Gender neutrality is fundamental to realising this future. Thank you. I await your reply soonest.