raspberrypi/piserver #71: You are wrong MAXNET: please to review your decision with following documentation reference:

arclite7, 2019-07-03:
You are wrong MAXNET: please to review your decision with following documentation reference:

   **### A minimal support for external LDAP/AD servers**

- Adds page to installation wizard to ask wheter user likes to
  create users with Piserver or use an external auth server.
- If user selects he wants to use external server Piserver will
  do a quick attempt to auto-detect the location of said server.

...as this was being responded to. If you are not willing to adjudicate reasonably or fairly on matters of GENDER NEUTRALITY it is reasonable for this matter to be escalated to include an equal number of responsible personnel with a "different gender" to that of your own as per European equality guidelines. 

**Women in decision-making: why it matters | European Institute ...**
[Search domain eige.europa.eu/news/women-decision-making-why-it-matters] https://eige.europa.eu/news/women-decision-making-why-it-matters
With the domain of power holding the lowest score in our Gender Equality Index, women remain under-represented in the kinds of decision-making positions that shape politics, economics and society. Yet gender equality is a founding value of the EU and improving it could lead to an increase in GDP of up to €3.15 trillion by 2050.

I certainly do not regard the issue of gender neutrality as trivial that it can be deprecated without reasonable response to the central issue! Without a single request for clarification?  The RPi business model approaches the most vulnerable people worldwide and all prospective references from that model must take responsibility for consolidating a gender neutral position in all aspects of their presentation to the RPi audience. Unless you would prefer to go head to head with a single gender supremacist perspective based upon a model of pre-existing social privilege? Please do explain your position for the record if you will? If you choose not, please be informed of my intention to make a formal complaint of your action since as I am a newcomer to Github you have failed, so far,  to make any aspect of this "negotiation" clear or informative. Gender neutrality may in fact be new to yourself. The RPi model is itself an attempt to make immediate access to cross referencing of factual information second nature to the forthcoming generations such that international collaboration will be the next revolution. Gender neutrality is fundamental to realising this future. Thank you. I await your reply soonest.

_Originally posted by @arclite7 in https://github.com/raspberrypi/piserver/issues/70#issuecomment-508017953_

ghollingworth, 2019-07-03:
@arclite7 

We at Raspberry Pi do try to make sure all our documentation is gender neutral, but sometimes things fall through the cracks.  If you've found a place where this is the case, just edit the documentation and submit a pull request.  Our team will copy edit and pull your change (or edit as appropriate).

@maxnet is correct that this project does not have documentation and therefore you have added this in the wrong place (it should have been brought up in the Documentation repo),  please raise a pull request there.

Thanks

Gordon

closed 2019-07-03 by ghollingworth

arclite7, 2019-07-03:
Dear ghollingworth, Again I am presenting an item from this authors project "PiServer", "a file containing text reference materials i.e. comments on the configuration and usage of various aspects of this project and its components".  To me this constitutes a document in reference to the functioning of this project and as such it is technically correct to refer to such as "Documentation" fit for purpose of identifying an accessible and incorrect non-gender-neutral item. The central issue here is GENDER NEUTRALITY and the impact of failing to maintain such gender neutrality is more important than semantic discussions of what constitutes a document and where such comments on documentation should be placed when in fact you are incorrectly referring to 'quote' non-existent documentation 'unquote' to which this alternate location for comment is moot. 

I assert the central issue of an inclusion of non-gender-neutral reference for resolution apparently in the correct location such that the continuity of its resolution is made absolutely clear due to the significance of this issue. In future all similar issues should also be dealt with similarly and would be more compact if respect for the presented issue was taken equally seriously by all concerned since sufficient evidence is also included to assist reasonable openness towards this issue? 

I thank you for the reference on the correct use of a "Pull request" as I find this most helpful and this will enable me to comply with conventions of Github in future. I find it hard to understand the referencing according to a file structure in Github such that only the title of the file containing the gender item has so far been provided. I am reliant upon Github itself providing the context to both the issue and comments?

Again I am appreciative of all help and support received and look forward to learning a great deal from this community. In order to learn it is necessary first to feel safe!

Best Regards

Alexandra Crawford

XECDesign, 2019-07-03:
Do you have a link to where the quoted text is found? I see it in a git commit message and a comment on an issue, but haven't found it anywhere else.

maxnet, 2019-07-03:
>Again I am presenting an item from this authors project "PiServer", "a file containing text reference 
>materials i.e. comments on the configuration and usage of various aspects of this project and its 
>components". 

Commit logs (like the one you quoted) are internal scribbling by developers describing what changes to code they made on a certain date.
Such changes tend to be in response to requests ("issues") from specific users, and are therefore highly contextual.
When a reference to a user is made, it could be that the developer had the person that requested the feature in mind, rather than making implications about what demographics Piserver users in general belong to.

This project intentionally did not produce any documentation meant for a wider audience, because its main author (me) is not a native English speaker.
While English is an easy enough language to understand and speak as a foreigner, writing proper documentation requires a different level.
So don't expect to find any real documentation in this repository, and if you see any shortcomings, it's caused by incompetence rather than malice. :-)