raspberrypi/piserver #72: User Login with Active Directory / LDAP setup

jonwitts, 2019-07-15:
Hi there,

I am thinking about moving my install of PiServer over to Active Directory for user logins as my students are forever forgetting their passwords on the Pi network... I realised that this will probably involve setting up the server from scratch again so I have been testing things out on a VM first... 

Using the stock PiServer from the ISO downloaded from the Raspberry Pi website I can only list users inside the Users OU.  I remembered seeing some activity on here about adding the ability to search for users outside of the Users OU in AD; so I downloaded and compiled the latest version of PiServer from here over the top. Interestingly they both reported the same version number...

My issue is that whilst I can now see all users from my AD network listed under users; I cannot seem to log into the PI with any of the accounts. I am just presented with "incorrect password". Is there some further configuration to do to be able to get my users logging in or are there some log files I could be looking at to find out what is going wrong?

Cheers,
Jon

maxnet, 2019-07-15:
There should not be any additional configuration necessary.
But the AD functionality has not been tested that extensively (lack of real school networks to play with), so there may well be bugs.

For testing purposes you may want to try logging in on the piserver computer itself with AD user details.
That should work as well, and it may be easier to view log files there, then on the Pi clients.

The LDAP configuration is stored in /etc/nslcd.conf (edit with "sudo leafpad /etc/nslcd.conf")
You may want to verify that the "base" DN looks right, and no cn=users leftovers from a previous installation are there.
And that the "map passwd uidNumber" line contains the SID of the domain your users are in (first part of user SID).
That is used to derive Unix user ID numbers from a Windows SID number.

By adding "log syslog debug" to the file, you can enable logging to /var/log/daemon.log
May be necessary to reboot for changes take effect.
Log output can be a bit cryptic though.

(If it turns out that any of the values in /etc/nslcd.conf need tweaking, you should after you have verified that logging in with an AD user on the piserver computers works properly, also alter /var/lib/piserver/os/your-raspbian-instance/etc/nslcd.conf for the clients)

jonwitts, 2019-07-17:
Sorry for the delay in getting back to you; I had to swap out a hard drive in my work PC and have been reinstalling everything!

So I can log onto the PiServer with my AD Credentials and everything in the /etc/nslcd.conf file seems sane. When looking at the /var/lib/piserver/os/Raspbian_Lite-2019-04-08/etc/nslcd.conf file though it looks as if it is not getting the values from my PiServer... Unless I am missing something? Both files copied below with passwords and domains edited...

Jon

---------------------
/etc/nslcd.conf (PiServer)
---------------------
# /etc/nslcd.conf
# nslcd configuration file. See nslcd.conf(5)
# for details.

# The user and group nslcd should run as.
uid nslcd
gid nslcd

# The location at which the LDAP server(s) should be reachable.
uri ldap://<DC IP Address>

# The search base that will be used for all queries.
base dc=school,dc=domain,dc=com

# The LDAP protocol version to use.
#ldap_version 3

# The DN to bind with for normal lookups.
binddn BindUser@school.domain.com
bindpw ReallyComplicatedPassword!

# The DN used for password modifications by root.
#rootpwmoddn cn=admin,dc=example,dc=com

# SSL options
#ssl off
#tls_reqcert never
tls_cacertfile /etc/ssl/certs/ca-certificates.crt

# The search scope.
#scope sub

referrals no
filter passwd (&(objectCategory=person)(objectClass=user)(!(UserAccountControl:1.2.840.113556.1.4.803:=2)))
map passwd uid              sAMAccountName
map passwd homeDirectory    "/home/$sAMAccountName"
map passwd gecos            displayName
map passwd loginShell       "/bin/bash"
map passwd gidNumber        "100"
map passwd uidNumber        objectSid:S-1-5-21-2224928304-2840136811-300653630

---------------------
/var/lib/piserver/os/Raspbian_Lite-2019-04-08/etc/nslcd.conf (Pi Client)
---------------------
# /etc/nslcd.conf
# nslcd configuration file. See nslcd.conf(5)
# for details.

# The user and group nslcd should run as.
uid nslcd
gid nslcd

# The location at which the LDAP server(s) should be reachable.
uri ldap://piserver

# The search base that will be used for all queries.
base dc=raspberrypi,dc=local

# The LDAP protocol version to use.
#ldap_version 3

# The DN to bind with for normal lookups.
#binddn cn=annonymous,dc=example,dc=net
#bindpw secret

# The DN used for password modifications by root.
#rootpwmoddn cn=admin,dc=example,dc=com

# SSL options
#ssl off
#tls_reqcert never
tls_cacertfile /etc/ssl/certs/ca-certificates.crt

# The search scope.
#scope sub

jonwitts, 2019-07-17:
I tried replacing the client OS nslcd.conf with the values in the PiServer nslcd.conf file but that did not seem to work either...

I will try adding debugging to it now and see if I can follow along with tail to see what errors we are getting...

Jon

jonwitts, 2019-07-17:
OK... All I can see that relates to the denied login in the log file are these two lines:

```
Jul 17 14:31:41 raspberry nslcd[634]: [a64e2a] DEBUG: connection from pid=416 uid=0 gid=0
Jul 17 14:31:41 raspberry nslcd[634]: [a64e2a] <passwd="�#001SUDO_USER="> request denied by validnames option
```

maxnet, 2019-07-17:
>When looking at the /var/lib/piserver/os/Raspbian_Lite-2019-04-08/etc/nslcd.conf file though it looks 
>as if it is not getting the values from my PiServer..

Strange.
Normally that file does is patched when the OS is first added to Piserver, by the same script that also borrows other settings from the host computer like timezone and keyboard settings etc.

/var/lib/piserver/settings.json does have ldapServerType, ldapUri, ldapDN, ldapUser, ldapPassword and ldapExtraConfig all set?

==

>OK... All I can see that relates to the denied login in the log file are these two lines:

That does is the log file on the client (and not on the server)?

(Easiest way to view log files on the client, is set a local root password, with "passwd root" in the chroot shell. And then login as root on the client Pi).

>Jul 17 14:31:41 raspberry

Is it an altered image?
Recall default hostname on the official images is raspberrypi

jonwitts, 2019-07-17:
> Strange.
> Normally that file does is patched when the OS is first added to Piserver, by the same script that also borrows other settings from the host computer like timezone and keyboard settings [...]
settings.json does have the ldapServer etc. set... Copied below:
```
{
    "domainSid": "S-1-5-21-2224928304-2840136811-300653630",
    "endIP": "<End IP Address>",
    "gateway": "",
    "installed": 1,
    "interface": "eth0",
    "ldapDN": "dc=school,dc=domain,dc=com",
    "ldapExtraConfig": "referrals no\nfilter passwd (&(objectCategory=person)(objectClass=user)(!(UserAccountControl:1.2.840.113556.1.4.803:=2)))\nmap passwd uid              sAMAccountName\nmap passwd homeDirectory    \"/home/$sAMAccountName\"\nmap passwd gecos            displayName\nmap passwd loginShell       \"/bin/bash\"\nmap passwd gidNumber        \"100\"\nmap passwd uidNumber        objectSid:S-1-5-21-2224928304-2840136811-300653630\n",
    "ldapPassword": "ReallyComplicatedPassword!",
    "ldapServerType": "ad",
    "ldapUri": "ldap://<DC IP Address>",
    "ldapUser": "BindUser@school.domain.com",
    "netmask": "255.255.255.0",
    "standaloneDhcpServer": 1,
    "startIP": "<Start IP Address>"
}
```
> That does is the log file on the client (and not on the server)?
>
> (Easiest way to view log files on the client, is set a local root password, with "passwd root" in the chroot shell. And then login [...]
The logs were on the server - I could find no logs on the client. I will set a root password and try again...
> > Jul 17 14:31:41 raspberry
> 
> Is it an altered image?
> Recall default hostname on the official images is raspberrypi
All installed from latest images from Foundation's website with no modifications made...

jonwitts, 2019-07-17:
> The logs were on the server - I could find no logs on the client. I will set a root password and try again...
Managed to log onto the Pi client as root but had no way of getting the log file copied! I have attached a picture of my screen...

[image]

maxnet, 2019-07-17:
Log shows that it is indeed not using the right nslcd.conf values.

>I tried replacing the client OS nslcd.conf with the values in the PiServer nslcd.conf file but that did not seem to work either...

Try just copying the file.
From a normal shell on the server:

```
sudo mv /var/lib/piserver/os/your-image/etc/nslcd.conf /var/lib/piserver/os/your-image/etc/nslcd.old 
sudo cp /etc/nslcd.conf /var/lib/piserver/os/your-image/etc/nslcd.conf
```

Also do have an idea about why the file was not updated.
Will push a fix for that shortly.

jonwitts, 2019-07-17:
OK; I thought I had done that but checking "less /var/log/daemon.log | grep nslcd" I noticed that the client could not contact the domain server... It helps if you configure the gateway in your DHCP server settings!

All working fine now with the conf file manually copied over and the gateway set in the PiServer settings...

maxnet, 2019-07-17:
Good to hear it works now.

Don't forget to disable the local root account again ("passwd -d root" in chroot shell)

(while it is handy to use for debugging, local user accounts like that should not be used in production for security reasons.
Given the hashed password is stored in /etc/shadow which is shared publicly by NFS within the network)

jonwitts, 2019-07-18:
> Given the hashed password is stored in /etc/shadow which is shared publicly by NFS within the network)

That's nice to know!

This is just a testing box which will be deleted once I am happy it's all working... Any news on a fix for why the settings are not copying across to the imported OS?

maxnet, 2019-07-18:
>Any news on a fix for why the settings are not copying across to the imported OS?

The linked commit should fix it: https://github.com/raspberrypi/piserver/commit/b5a500767b1f3e1ab8ec20c11e8e9024d3a54e54
(It's the script that is run after you add a new OS to Piserver. So commit will not edit the files within your existing operating systems)

jonwitts, 2019-07-18:
Excellent! I shall install another testing Pi Server, recompile and get back to you.

Thanks :-D

jonwitts, 2019-07-30:
This is all working beautifully now! Thanks :-D

closed 2019-07-30 by jonwitts