raspberrypi/piserver #73: Filter LDAP Server logins by group memberships

jonwitts, 2019-07-17:
Hi there,

Still testing the LDAP / Active Directory linking with PiServer and I have a request!

It would be great if there were a way in the setup GUI to add a group that users should members of if we want them to log in. I can't think of many school setups where all domain users would need to log in to the PiServer network...

I have done some testing on our setup and if you adjust this line in /etc/nslcd.conf:
`filter passwd (&(objectCategory=person)(objectClass=user)(!(UserAccountControl:1.2.840.113556.1.4.803:=2)))`
to this (changing it to your CN for your group you want to check against of course!):
`filter passwd (&(objectCategory=person)(objectClass=user)(memberof:1.2.840.113556.1.4.1941:=CN=PiServer Users,OU=Groups,DC=school,DC=domain,DC=com)(!(UserAccountControl:1.2.840.113556.1.4.803:=2)))`

then it will only allow users who are members of the group to log in (both direct and nested members work with this filter). I haven't checked this against the pi client yet due to Issue #72 but should think it will work fine once I can get that part working!

This would be a really nice addition to the setup... :-D

maxnet, 2019-07-17:
Problem is that there is more than one way users can be grouped in LDAP.
E.g. some organizations like to only group users in containers ("CN=user,OU=Some group,DC=domain,DC=com"), in which case you would not specify a filter, but just provide a more exact baseDN value.

May be a bit of a challenge to add an option in the GUI to handle all that, without scaring away the less technical users.

jonwitts, 2019-07-18:
How about the ability to edit the default filter with a caveat of "make sure you know what you are doing!"?

That way at install more technical users could adjust the filter how they needed and have it automatically copied to all future OS they import etc...

maxnet, 2019-07-18:
Added a simple box to select group during initial installation.

jonwitts, 2019-07-19:
That's great! I am on leave for a week now but I will test this and issue #72 when I get back from holiday at the end of the month.

Thanks again :-)

jonwitts, 2019-07-30:
This is great - the list of available CNs and groups was a little long to search through but better than having a user enter an non-existent group or CN for those settings!

Thanks again :-D

closed 2019-07-30 by jonwitts