raspberrypi/piserver #77: Epoptes - Monitoring clients visually

marc-odp, 2019-08-02:
Hi,
I have a demand of a teacher to be able to "see" remotely what students are doing (in case they go on Youtube instead of doing their work...)
I found Epoptes installation doc here : http://www.epoptes.org/installation
In Piserver, in the "Clients" window, it is possible to open a terminal window directly in the operating system. Can I install the client of Epoptes in this ? Like described : 
apt-get install --install-recommends epoptes-client
epoptes-client -c             # Fetches the OpenSSL certificate from the server
And putting a line like 1.2.3.4 server to /etc/hosts
Sorry if this is a stupid question but I prefer to ask before messing up with my working Piserver installation. :-)
Thank you.
-marc-

maxnet, 2019-08-02:
>In Piserver, in the "Clients" window, it is possible to open a terminal window directly in the operating system.

Go to the "software" tab instead of clients, select the operating system you wish to make the changes, and press the "shell" button there.

>And putting a line like 1.2.3.4 server to /etc/hosts

The server is available under the hostname piserver
Easiest is not to mess with /etc/hosts, but do:

```
echo "SERVER=piserver" > /etc/default/epoptes-client
```

Not sure if the Eptoptes version you get with apt-get works properly enough.
It may be necessary to compile it from source instead.
List of issues it had previously: https://github.com/raspberrypi/piserver/issues/8

marc-odp, 2019-08-02:
Thank you. I don't understand how I missed that closed issue !