raspberrypi/piserver #79: Installing a new custom OS

sdnei-avelar, 2019-08-30:
Hi Guys,

I have a PI 3+ B from where I want to export a template to be used by all PIs in my lab environment.
What would be correct procedure to import it to Pi Server as a tar.xz file?
Do I only have to compress an exported img file into tar.xz, then install it?
The plan I'm going to try is:
1 . Export the SD card into IMG using DD;
2. Compress the IMG file into tar.xz
3. Install the new tar.xz file into PI Server.

Any input is highly appreciated.

Cheers,

Sdnei

maxnet, 2019-08-31:
>Do I only have to compress an exported img file into tar.xz, then install it?

Technically, yes.

However is your custom OS already capable of using a read-only root file system?
As in, if you add "ro" to the list of options in cmdline.txt, and boot it from SD card, everything works as you want it to be?

sdnei-avelar, 2019-09-02:
Hi Maxnet,

Thanks for your reply.

The answer is no, I haven't changed cmdline.txt.
I haven't done anything to the PI's FS yet, apart from customising the OS with my lab requirements, such as changing usernames and enabling services such as SSH and XRDP.
Now I want to extract an image from this Pi, then upload it into the PiServer that then can be leveraged by all Pis in my lab via network boot.
I've tested with the default images provided via the repositories and everything works fine.
Adding more information, my base template OS is based on Ubuntu Mate 18.
I also tried compressing a default Ubuntu Mate img file into tar.xz, but it didnt work at all :)