raspberrypi/piserver #82: Raspian Buster Will not boot with Multiple HDs attached

gorky, 2019-09-11:
Setup:
Raspberry PI B
2 HDs
* 750GB Drive for OS and general data.  This drive has a Boot sector.
* 2TB Drive -- NO Boot sector.
* Release: 2019-07-10

Previous version (Stretch?) supported booting with two HDs attached.  Current version will boot with just the 750GB Boot/OS drive, but will fail to boot when both drives are attached.  I've tried attaching to OS Drive directly to the PI w/o a powered Hub.  Surprisingly, it works -- so long as 2nd drive is not plugged in anywhere during boot.  Plug in 2nd drive, and PI doesn't boot. No Raspberries on the screen even.  One drive plugged in (either direct or via powered USB Hub), and system boots all the way.

This includes kernel 4.99.66-v7+.

HDs are being used rather than SD cards as the logging wears out the SD cards fairly quickly.

gorky, 2019-09-11:
4.19.57-v7+ is also impacted.

maxnet, 2019-09-13:
This repository is only about the piserver software.
A GUI for managing network booting Pi.

Your USB disk boot issue probably belongs instead at: https://github.com/raspberrypi/firmware

closed 2019-09-13 by maxnet