raspberrypi/piserver #85: Using PiServer for home projects

HarvsG, 2019-09-16:
Hi when I read the PiServer blog [post](https://www.raspberrypi.org/blog/piserver/) I got very excited by this:
  

> Build a home server
PiServer could be used in the home to serve file systems for all Raspberry Pis around the house — either a single common Raspbian file system for all Pis or a different operating system for each. Hopefully, our extensive OS suppliers will provide suitable build files in future.

My vision was this - to host all my images/OSs on a central piserver host and then disk them out to a rack of RaspberryPis. I could then use piserver to control which pi boots which OS. This much I can do with piserver. However it is hard to to any projects of consequence as I only have write access to `/home/user`.

Is there an experimental way to allow write access to the root/image (I would make sure each image is only booted by one client at a time). Or better still for piserver to store changes in `/var/lib/piserver/nfs/<pi's-mac-addr>/<os-name>/` then load these changes/files on boot.

maxnet, 2019-09-16:
You can have a different OS image for each Pi (if you compile piserver from source, there is a "clone" button in the Piserver GUI you could use to make a copy)
And can then change /etc/exports so read/write NFS access is given.

However do be aware that this is NOT secure.
Anyone in your network, will have write access to any OS image.
No authentication whatsover.

HarvsG, 2019-09-16:
Thank you so much for your quick reply

Ok so I did the following
    - on the piserver `sudo nano /etc/exports`
    - in `/var/lib/piserver/os *(ro,no_subtree_check,no_root_squash,fsid=1055)` changed `ro` to `rw`
    - `sudo exportfs -ra`
    - selected each OS, open the shell in piserver to chrooted into it them `nano /etc/sudoers`, added my user `HarvsG` as a sudo.

Then network booted, logged in as `HarvsG` and ran `sudo apt-get update`. However I get read-only filesystem errors.

maxnet, 2019-09-16:
Does it work better if you add " rw" to the line of existing options in /boot/cmdline.txt of the OS image?

HarvsG, 2019-09-16:
I added ` rw` to the end of cmdline.txt adding `,rw` caused a kernal panic. 

However now I can't log in. It just keeps telling the password is wrong. Could this have affected the authentication?

maxnet, 2019-09-16:
>This apears to be a separate issue

Do keep in mind that most of Linux is CaSe-SeNsItive.
So try to keep things lower case and avoid usernames like HarvsG.

You have multiple options regarding authentication.
Can use piserver's user management, in which the home folder will be central, and shared with all images. 
Avoid logging in with same username on multiple Pi at the same time. As not all applications will be expecting that.

Or you can have home folders specific to each image.
In that case do not use Piserver's (ldap) user management, but create users with adduser and similar command line commands in the shell.

HarvsG, 2019-09-16:
> Or you can have home folders specific to each image.
In that case do not use Piserver's (ldap) user management, but create users with adduser and similar command line commands in the shell.

I think this is the best option as my use case, how would I got about doing this?

Thank you so much for your help

maxnet, 2019-09-16:
>I think this is the best option as my use case, how would I got about doing this?

- Open a shell in Piserver GUI.
- "adduser username"

Can also set a root password with "passwd".
Login on the client as root, and add other users there.

HarvsG, 2019-09-16:
Thank you. I got this working - this could be game changing! Thank you for all your help. I'm gonna write up a tutorial.

maxnet, 2019-09-16:
Good to hear it works for you.

>I'm gonna write up a tutorial.

Do make sure you mention the security implications of setups like this.
It is fine to do it like this in your private home network, but this kind of configuration is not suitable for say schools.

Both because of everything being read-write through NFS for every computer in the network.
And because when you create local users like this, the password hash of the user ends up in /etc/shadow, which is shared publicly over NFS.

(When using Piserver's LDAP authentication this is not the case. Server has a healthy distrust of clients. Only gives access to a user's home folder, after client sends password to server)

HarvsG, 2019-09-16:
Thank you for this, I wonder if there is a way to overcome some of these security issues. I would have thought that specifing a more secure address range `/var/lib/piserver/os *(ro,no_subtree_check,no_root_squash,fsid=1055)` here or adding some sort of user authentication the the nfs.

Not sure how to solve the /etc/shadow issue.

Developing my tutorial here:
https://gist.github.com/HarvsG/7d1674e7aadc719d4a82da26731a1501

HarvsG, 2019-09-16:
one last question @maxnet will I be able to run piserver host on a raspberry pi 4 rather than using a x86 device?

closed 2019-09-16 by HarvsG

maxnet, 2019-09-16:
>one last question @maxnet will I be able to run piserver host on a raspberry pi 4 rather than using a 
>x86 device?

Technically yes.
(Although using x86 is typically better storage wise)

HarvsG, 2019-09-16:
Great I'm going to slap a USB 3 SSD onto a raspi 4 a use it as a master pi dishing out gigabit NFS filessytems! This is so much nicer than messing around with berryboot and iSCSI.