raspberrypi/piserver #86: Running XRDP on a custom OS image

sdnei-avelar, 2019-09-16:
Hi Guys,

I've been trying to setup a custom image based on Raspbian, which took me quite some time, however it is working now (buster only worked after blacklisting some kernel modules).
The scenario here is that all my users will access the RPIs via RDP since this is for a remote lab environment.
I installed XRDP on the original Pi and it works fine, however the XRDP service refuses to start up after the image is converted into the Pi server. Interesting enough VNC works fine.
As simple test, I exported the file system as 'ro' and changed the cmdline.txt to 'ro' as well, then from the network booted Pi I removed and reinstalled XRDP and service went up, but then when I made the file system 'ro' again, the service went down.
It is clear to me that XRDP does not like read-only file system and I'm still digging up for a workable solution.

Has anyone gone through this or have any ideas?

Cheers!

maxnet, 2019-09-24:
Problem is that xrdp expects `/var/log/xrdp.log` to already exists and be owned by the xrdp user.
In piserver setups /var/log lives in tmpfs (ramdisk) and is empty on each boot.

A workaround would be opening `/etc/xrdp/xrdp.ini` (e.g. in leafpad in the piserver chroot shell), and replacing:

`LogFile=xrdp.log`
with:
`LogFile=/dev/null`

(or add commands to the systemd unit that touch and chown /var/log/xrdp.log to the right user, prior to starting the main executable)

sdnei-avelar, 2019-09-27:
Changing LogFile to /dev/null worked like charm.

Thank you!