raspberrypi/piserver #88: Ethernet connection dropping

lgerster, 2019-09-25:
I have an issue with a headless raspberry pi booted using PiServer, after a period of about 24h it no longer visible on the network.
It reconnects properly after power cycling the device, but given the lack of local storage, there are no log files indicating when and why connection was lost. Since this is the only raspberry showing this behavior on the network, I assume it is related to PiServer.

I would appreciate any help in trying to troubleshoot this issue.

maxnet, 2019-09-25:
Is the Pi in the vicinity of another Pi/device?
If so I suggest you connect it to serial console. 
Put `enable_uart=1` in config.txt, `console=ttyAMA0,115200` in cmdline.txt, `ForwardToConsole=yes` in `/etc/systemd/journald.conf`

maxnet, 2019-09-25:
BTW does your DHCP server issue 24 hour leases?
Renewing it should work properly if it gets same IP-address as before, but not sure how well clients deal with an IP change.

lgerster, 2019-09-25:
@maxnet  where should these textfiles be located? 
The DHCP server does issue a static IP to the device

maxnet, 2019-09-25:
Either go to the piserver GUI, tab "software", choose shell, start "leafpad" there to edit the files.
Or edit it on the host system, in which case they will be under /var/lib/piserver/os/<name of operating system>

cmdline.txt and config.txt live under /boot

lgerster, 2019-10-02:
I do have another raspberry nearby, but it is headless as well, and I cannot really get at any ports but USB and ethernet.

maxnet, 2019-10-02:
>I cannot really get at any ports but USB and ethernet.

Well, I am afraid that is not going to work.
Cannot debug why Ethernet is failing through Ethernet...

lgerster, 2019-10-08:
I did plug in a raspberry where I can connect a monitor, and it doesn't seem to be the network conenctions fault, but the whole raspberry that is crashing. As it still displays the console, but doesn't react to any commands anymore

lgerster, 2019-10-31:
I have a suspicion it might be related to the readonly filesystem with dhcp leases expiring, but I am not sure how to fix it.
manually trying to refresh the dhcp lease gives me the following errormessage:
sudo dhclient eth0
/sbin/dhclient-script: 46: /etc/dhcp/dhclient-enter-hooks.d/samba: cannot create /etc/samba/dhcp.conf.new: Read-only file system
mv: cannot stat '/etc/samba/dhcp.conf.new': No such file or directory
RTNETLINK answers: File exists

lgerster, 2019-11-08:
The last lines of the syslog before the system crashes are:
dhcp[476]: eth0: NAK: from 192.168.0.1
dhcp[476]: eth0: deleting default route via 192.168.0.1
dhcp[476]: eth0: deleting route to 192.168.0.0/24
avahi-daemon[324]: Withdrawing address record for 192.168.0.243 on eth0
avahi-daemon[324]: Leaving mDNS multicast group on interface eth0. IPv4 with address 192.168.0.243.
avahi-daemon[324]: Interface eth0.IPv4 no longer relevant for mDNS.

maxnet, 2019-11-08:
>dhcp[476]: eth0: NAK: from 192.168.0.1

DHCP server says IP may no longer be used?

Got logs from the DHCP server side of things?

lgerster, 2019-11-11:
Unfortunately there is nothing in the DHCP log at that time, but then the NETGEAR logs are not very detailed.

maxnet, 2019-11-11:
Is there any chance your Netgear was rebooted/power cycled?

Most of those toy routers, do not store the DHCP leases table on persistent storage, but in memory.
Meaning that if you reboot them, it no longer knows which IP address was assigned to which MAC, and do not allow the client to renew its current IP.

You may be able to workaround that by entering static MAC<->IP mappings for the client in the router's interface, if it offers an option for that.

lgerster, 2019-11-11:
The raspberries have static IPs assigned on the router to their MAC addresses. The router also doesn't show anything in its log at that time.

maxnet, 2019-11-11:
What are the syslog lines above the NAK?
192.168.0.1 do is the router that is supposed to play DHCP server and not another device in the network?

lgerster, 2019-11-12:
There is nothing DHCP related in the log before, only some rngd messages and CRON running earlier.
Yes, 192.168.0.1 is the router, which is supposed to be the DHCP server