raspberrypi/piserver #90: network printing is very slow

Gio1998, 2019-10-07:
I can't figure out what problem when sent job snmp takes too long

maxnet, 2019-10-07:
snmp?

Did you install CUPS on the piserver server, like I suggested in your previous issue #80 ?
Or are you network printing in any other way?

Gio1998, 2019-10-07:
when I try to print os which is on sd card it works but on piserver os when I sent Printer job to a printer it take too long time to communicate each other with  SNMP protocol

maxnet, 2019-10-07:
The Simple Network Management Protocol (SNMP) is typically used for managing devices like Ethernet switches, and the monitoring resource use of computers and devices.
It's what applications like nagios and observium use to gather statistics about for example data traffic usage, to draw pretty graphs of.
So are you sure it is not a different protocol?

Do you have a screenshot/photo of the actual error?

If you made any folder writable on the piserver (like you said you intended to in the other topic), does the problem also exist if only a single Pi is booted?

Gio1998, 2019-10-08:
I am attaching printers each host individuali with lpadmin and when i send print query from host it is very slow

maxnet, 2019-10-08:
>I am attaching printers each host individuali with lpadmin

Why individually, and not to the piserver server instead, like suggested?
https://github.com/raspberrypi/piserver/issues/80

lpadmin may have a problem storing the settings and print job queue on a read-only file system.
Or did you make more adjustments then just running lpadmin?

Gio1998, 2019-10-12:
i add in fstab  /etc/cups as additional partition add when booting rc.local copy cups from /usr/local/ to   /etc/cups and then i can change and add files in /etc/cups

maxnet, 2019-10-12:
But doesn't cups normally write to other locations such as /var/spool/cups /var/cache/cups /var/log/cache ?
Or did you override those locations in the config?

Gio1998, 2019-10-22:
thank it helped me a lot

closed 2020-02-18 by Gio1998