raspberrypi/piserver #91: Pi4 Dual HDMI Output from PiServer

jonwitts, 2019-10-08:
Hi there,

I have my Pi 4 devices booting from a Buster image on my PiServer. The RPi that I have on the teacher's desk is using dual HDMI outputs; one to the monitor and one to the projector in the classroom. I would like to have this running in an extended desktop mode, but it will only seem to duplicate the display. 

When I boot this RPi with a Buster SD Card, rather than from the PiServer it defaults to extended desktop mode. 

Is there any way I can get a Buster image on the PiServer to default to extended desktop mode if it detects both HDMI are connected?

Thanks,
Jon

maxnet, 2019-10-23:
As a workaround for now go to chroot shell and excute:

``
rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf
``

Suspect the script that patches up the X configuration depending on wheter or not fkms overlay is active, does not likes read-only file systems.

jonwitts, 2020-01-10:
Hi there,

Just gotten round to testing this in my lab and I am happy to say that removing (I just renamed it to be on the safe side!) works a treat.

I now have dual screen from my Pi at the front of the class (connected to my monitor and classroom projector) and all students computers are running nicely with single screen.

Thanks.

closed 2020-01-10 by jonwitts