raspberrypi/piserver #94: RPi4 Support on PiServer version 0.5

jonwitts, 2019-12-16:
Hi there;

With the release of the Buster version of the x86 Raspberry Pi Desktop: https://www.raspberrypi.org/downloads/raspberry-pi-desktop/ has the ability to add Raspberry Pi 4 been included in  the software? 

I am keen to test an "official" release that allows RPi 4 to be added and managed from PiServer rather than my somewhat hacked way of doing things at the moment! https://www.jonwitts.co.uk/archives/1222

I have also seen that network booting of the RPi4 is now being tested / in beta...

Are there plans to get this incorporated into PiServer if it has not already been?

Thanks,
Jon

maxnet, 2019-12-16:
As far as the Piserver side of things is concerned Pi 4 works, yes.

They do not come with network booting enabled by default though.
And I personally think the currently recommend way of updating EEPROM using a SD card with Raspbian (like documented at: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md )  is a bit overly complicated if you want to enable it on multiple Pi.

Would perhaps be easier if a ready-made tiny .zip file or image with just the bits needed to update EEPROM with the new firmware and settings was released. (not using a full Raspbian installation for that, but just using the recovery.bin method).

jonwitts, 2019-12-16:
Thanks for the reply... Yes; I would agree that does seem overly complicated! 

Once I have copied and edited the bootconf.txt could I use the same one to just apply the change to all my Pi 4, or would I need to extract it and edit it for each one?

Let me know if this isn't the right place for such question!

maxnet, 2019-12-16:
>Once I have copied and edited the bootconf.txt could I use the same one to just apply the change to 
>all my Pi 4, or would I need to extract it and edit it for each one?

- If you follow the procedure once you will have an pieeprom-new.bin at the end.
- Take that file, rename it to pieeprom.bin, and put it on a blank FAT formatted SD card.
- Now steal recovery.bin from the emergency EEPROM recovery stuff: https://static.raspberrypi.org/files/rpi-boot-eeprom-recovery-2019-09-10-vl805-000137ab.zip

Think you should be able to use that SD card to flash all your Pi 4.

>Let me know if this isn't the right place for such question!

Yes and no.
Asking in rpi-eeprom will probably give you a more official answer, what the proper way is.
That does not mean, my solutions do not work. :-)

jonwitts, 2019-12-16:
Thanks; I will give it a go later this week hopefully and let you know how it goes :-D

jonwitts, 2020-01-15:
Hi that pretty much worked... The only bit I had to add was creating a pieeprom.sum file using:
`sha256sum pieeprom.bin > pieeprom.sum`

This was answered over on the RPi-EEPROM repo for me: https://github.com/raspberrypi/rpi-eeprom/issues/78

Thanks - I now have a handful of RPi4 network booting from my PiServer! I am going to make my way round them all in the next couple of days and can once again do away with the SD Cards completely!

closed 2020-01-15 by jonwitts

MilesAheadToo, 2021-11-12:
Is it possible to run PiServer on an RPi4?

maxnet, 2021-11-17:
>Is it possible to run PiServer on an RPi4?

Technically, yes.
Would still recommend Intel/AMD for better storage options, though.