raspberrypi/piserver #96: Running PiServer on Raspberry Pi 4

locutusofsand, 2020-01-24:
I have PiServer running on an older server and I wanted to speed test it against running on a Raspberry Pi 4. However, I have come to find that PiServer does not appear to work in Raspbian. 

When I launch PiServer I get the Introduction screen and I can go through all of the steps of finding clients, setting up users and choosing OS, and downloading software. All of that works fine, but when it finishes the download, it quits the Introduction wizard, asks you to login as root as expected and just starts the introduction screen again. You never get to the point where you can manage any of the clients. I was able to replicate this behavior on 2 different installs of Raspbian.

locutusofsand, 2020-01-28:
Digging through the code a little bit I see that there is a setting that populates on a successful run of the install wizard. I checked the settings.json file on the system and only the interface and ldap password are populated, so onInstallationSuccess is never run from installwizard.cpp. It appears that the downloadextractthread does seem to succeed as I do get a os folder that does have files in it, but I would assume that the dependenciesinstallthread is what is failing. 

Is there a log that I can verify that at?

XECDesign, 2020-02-06:
I'm seeing this as well. I managed to make it through the first run screen once, but it still sometimes crashes when installing an OS (at the end, after it's extracted). All I see in the terminal is 'Bus error'.

paulatdunedin, 2020-02-12:
On the Raspberry Pi 4 the piserver desktop menu item has the the property:
  Exec=sh -c "sudo piserver || pkexec piserver"
I think this means that if 'sudo piserver' doesn't succeed then execute 'pkexec piserver'.
Logged in as pi, I tried opening a cmd window and simply entering 'sudo piserver' which worked.
Also 'pkexec piserver' worked too.
I got the idea having looked at the piserver desktop menu item on a virtual PC build which reads:
  gksudo piserver

maxnet, 2020-02-19:
Any more luck with 47fe8768a56c9ba97018e2662a3588487d3e66cb ?

XECDesign, 2020-02-19:
I've just done a few installs without the issue popping up again.

Any pending changes, or should we get a 0.6 out?

paulatdunedin, 2020-02-19:
I compiled 47fe876 on a Pi 4 and it installs fine. Doesn't exit when I select 'Authenticate users against an existing server' option. Quick tests with local account management setup and with Windows Active Directory account management seem to work fine.

maxnet, 2020-02-19:
No pending changes.
(Although some folks on the forum reported some problems with shared folders.
It does not seem so reproducible)

locutusofsand, 2020-02-20:
I can confirm that 0.6 solves the problem. Thanks everyone for their help!

closed 2020-02-20 by locutusofsand