raspberrypi/piserver #100: Can I use bluetooth

gits00, 2020-03-23:
I wish to use a network booted RPi as a data collector from Bluetooth devices. I was able to boot a pi 3B+ with a minimal image of Rasbian buster lite. But I am unable to get the Bluetooth working probably because of the security risk which prevents Bluetooth to not work unless started using "sudo bluetoothctl" (https://github.com/DrRowland/RPi-Bluetooth-Console/blob/master/setup.sh). But I feel there would be some workaround to get this because a networked sensor array is one of the main use cases of piserver.

maxnet, 2020-03-23:
Have not tried Bluetooth.
And I do not know whether or not that wouldn't like to write information like paring data to locations that are read-only on a default piserver installation.

But generally speaking:

- Nothing prevents you from running scripts as root. e.g. by writing a systemd .service file that starts said script, or putting the commands in /etc/rc.local
- I very much doubt you need to be root to execute bluetoothctl in the first place.
People often use "sudo" as a quick fix to get around any permission problem.
But I would try if adding the user you are currently using to the bluetooth group doesn't work first...

gits00, 2020-03-26:
@maxnet, Thanks. 

-  I tried running "bluetoothctl" as the user. It gets hung. 

- If "sudo bluetoothctl" is run, I get error the user not in the sudoers file. 

- On trying to add the user in bluetooth group in chroot shell of piserver got error -  user doesn't exists.

Is there anything I am missing here? Thank you.

maxnet, 2020-03-26:
If you want to add network users (created in Piserver) to a group, easiest is to do so by executing "leafpad /etc/security/group.conf" and adding a line among the lines of:

```
# Extra groups for all users logged in to local console
login;tty*;*;Al0000-2400;bluetooth, sudo
```

Might need to use something different than "login" if connecting through ssh, serial console or other means.
Or specify individual users instead:

```
# Give myuser extra groups regardless how he logged in, at all times of the day
*;*;myuser;Al0000-2400;bluetooth, sudo
```

gits00, 2020-03-28:
Thanks @maxnet. I was able to get bluetooth working in interactive session though a couple of times I see UART initialization timed out in "/var/log/syslog" . In such cases "bluetoothctl" got hung. 
At other times bluetooth worked. The changes needed were - 
- Changes suggested by you in "/etc/security/group.conf" (Thanks a lot for this one. Good learning for me) 
- Commenting out these lines in /boot/config.txt
#dtoverlay=pi3-disable-wifi 
#dtoverlay=sdtweak,poll_once 
(Probably because WiFi and Bluetooth radios are the same)
- Enabled line - "sudo	ALL=(ALL:ALL) ALL"  in "/etc/sudoers"
- Removed "nolink" from "/etc/dhcpcd.conf". Not sure if it was necessary but config.txt mentions it.

Do I need to create a systemd service here? I see bluetooth service getting started at boot time. Looking around found that "BLUETOOTH_ENABLED=1"  in "/etc/default/bluetooth". I am pretty new to using bluetooth and Raspberry Pi so not really sure.

maxnet, 2020-03-28:
>Do I need to create a systemd service here? I see bluetooth service getting started at boot time.

Well, you mentioned wanting to use it as data collector.
If by that you mean that multiple Pi collect data from bluetooth sensors, and phone it back home to some central server, then the daemon that does the data collection typically would run as a systemd service, yes.
That's not Piserver specific though.

gits00, 2020-03-30:
Oh ok, I got it now. Thanks @maxnet !

closed 2020-03-30 by gits00